volkszaehler/volkszaehler.org #210: Feature request: Date range selector in Frontend

nyphis, 2015-01-03:
I would like to see a more advanced way to select the time range shown in the graph.

In my eyes the date range shown as heading could be a little more "interactive". 
If you click a date (start date or end date) a date selector div (calendar) shows up to enter the exact date range to be shown. 

I've seen something like this in Google Analytics (see attachment) and really liked it.
This would make things easier to select a complete day or month e.g. to get the overall cost of my heat pump.

[image]

andig, 2015-01-05:
If interested start here http://jqueryui.com/datepicker/#date-range

andig, 2015-02-21:
@nyphis which app did you take the screenshot from?

nyphis, 2015-02-21:
Google Analytics

 Frontend (#210)

https://github.com/volkszaehler/volkszaehler.org/issues/210#issuecomment-75369609

andig, 2016-06-25:
Tipp: um auf Tag/Monat/Jahr zu zoomen einfach einmal dafür sorgen dass das Chart den aktuellen Zeitpunkt nicht umfasst- dann werden jeweils komplette Intervalle selektiert.

closed 2016-06-25 by andig