volkszaehler/volkszaehler.org #216: zoom presets should keep time aligned to "now"

r00t-, 2015-01-08:
... at least / especially if the "now" button was used before.
this bugs me too, and it's old:

http://volkszaehler.org/pipermail/volkszaehler-dev/2014-May/003592.html
> Einen Verbesserungsvorschlag fürs Frontend würde ich auch gerne Vorschlagen:
> 
> Beim Zoom auf die Stunde nach Verwendung von "jetzt" fände ich es
> angenehm, wenn auf die aktuelle Stunde gezoomt würde und nicht die
> mittlere Position der Darstellung
> genommen wird, was im Test bei mir gestern regelmässig 12 h früher war.
> 
> Wo ist der richtige Ort für solche Change Requests/Bug Reports?

andig, 2015-01-08:
+1 leider gabs im Code keine offensichtliche einfache Möglichkeit dafür.

jahir, 2015-01-08:
Sollte recht einfach sein. Zoom+ macht das ja auch schon. Dabei wird das Flag vz.wui.tmaxnow ausgewertet (siehe https://github.com/volkszaehler/volkszaehler.org/blob/master/htdocs/frontend/javascripts/wui.js#L562), das kann auch für die anderen Zoomstufen übernommen werden.

Allerdings wäre es evt. sinnvoller, rechts eine checkbox für diesen "sticky" mode zu setzen und das Verhalten darüber zu steuern. DIe aktuelle Logik in Zeile 430 ist auch eher optimisch (gelinde gesagt) und könnte auch mal gerade gezogen werden.

PR #218 (2015-01-10): Fix zoom in
Fix #216, replace #217.
closed 2015-01-10 by andig