volkszaehler/volkszaehler.org #227: composer install installs 23meg of jpgraph docs that no user needs

r00t-, 2015-01-20:
root@raspberrypi:~# du -sch /srv/volkszaehler.org/vendor/jpgraph/jpgraph/lib/JpGraph/docs
23M     /srv/volkszaehler.org/vendor/jpgraph/jpgraph/lib/JpGraph/docs

that (plus probably the copies in .git) accounts for at last a third of the vz.org installation footprint...
does this really have to be installed from git, given that it's used so rarely?

jahir, 2015-01-20:
Surely not. How can we prevent composer from installing it?

andig, 2015-01-21:
Why is that a Problem? Every other package is installed from git, too?

Viele Grüße, Andreas

andig, 2015-01-21:
Ups, I see. I'll create a fork and remove the docs.

andig, 2015-01-21:
Opened issue with owner: https://github.com/ztec/JpGraph/issues/7

Meanwhile forged and updated jpgraph: https://github.com/volkszaehler/JpGraph/commits/master

Updated dependencies: https://github.com/volkszaehler/volkszaehler.org/pull/229

r00t-, 2015-01-21:
let me try to clarify:
-  this is for a very rarely used feature, that most users don't know of and will never use
  (rendering graphs in the middleware instead of frontend)
- imho this neither justifies downloding many megabytes, nor a managing a fork
- does deleting files from git really reduce the download volume?
  matter of fact, it does not:
```
  $ git clone https://github.com/ztec/JpGraph JpGraph_ztec
  Receiving objects: 100% (2522/2522), 12.77 MiB | 2.94 MiB/s, done.
  $ git clone https://github.com/volkszaehler/JpGraph JpGraph_vz
  Receiving objects: 100% (2528/2528), 12.77 MiB | 4.47 MiB/s, done.  <---
  18M	JpGraph_vz
  46M	JpGraph_ztec
```

isn't there a composer target that just downloads a release tarball?
and even that could be made optional

also: from what i can tell, that ztec version isn't even the official one...?
what makes it better, why do we use it over the one from jpgraph.net?
(other than that composer prefers pulling stuff from VCSes?)

andig, 2015-01-21:
> isn't there a composer target that just downloads a release tarball?

Nicht bekannt- und davon werden die Daten auch nicht weniger.

> also: from what i can tell, that ztec version isn't even the official one...?

es gibt kein aktuelles Composer Package da das Projekt tot ist.

> what makes it better, why do we use it over the one from jpgraph.net?

Weil es funktioniert.

Anders gesagt: Du kannst aus unserem JP Repo gerne alles rausschmeissen was Dich stört, ich weiss aber nicht was davon gebraucht wird...

andig, 2015-01-22:
> 18M JpGraph_vz

Argh, natürlich- die Historie bleibt ja drin. Versuchs bitte nochmal, hab das Repo neu initialisiert.

andig, 2015-01-24:
Wir könnten auch auf phantomjs umstellen. Das muss unter Windows aber von Hand installiert werden?

jahir, 2015-01-25:
Äh, phantomjs ist doch ein kompletter Browser, nur headless. Größenmäßig werden wir dadurch wohl eher nichts gewinnen...

andig, 2015-01-25:
Für Windows 8MB, aber stimmt. Dafür gibts dann die gleichen schicken Charts wie im Frontend.

andig, 2015-02-19:
**ping**

wir könnens auch einfach sein lassen, aber wenn ich mir die Mühe schon mache wäre Feedback wirklich das Mindeste :/

r00t-, 2015-02-19:
i just noticed the problem and thought i'd report it.
i'm fine with any fix really.
so just merge https://github.com/volkszaehler/volkszaehler.org/pull/234 if you like

if you want my personal preference, then i'd make installing jpgraph optional and off by default. the feature is rarely used anyway.

andig, 2015-02-20:
Closing as wfm.

closed 2015-02-20 by andig