volkszaehler/volkszaehler.org #238: Use Highstock to display graph

Xento, 2015-02-01:
Is it possible, that you could switch to Highstock for displaying the graph?
http://www.highcharts.com/stock/demo
It supports touch devices, zooming, range selection and much more.

adlerweb, 2015-02-01:
Commercial software - don't this this would be a good choice…

Xento, 2015-02-02:
It is free for non commercial projects.

adlerweb, 2015-02-02:
Volkszähler is not solely used in non-commercial environments so implementing it as a default could lead to involuntary license violations. Also the GPLv3 used here doesn't tolerate an NC-part so I don't think it would even be possible to create a bundle containing both codes.

jahir, 2015-02-02:
One could code a new frontend with Highstock, but (as @adlerweb wrote) they would not probably not be allowed to re-use the current frontend code for it. Maybe there are ways to work around that (e.g. just have the API calls in the frontend code and let the user download Highstock separately), but I don't know if that's ok. Anyway, good luck with finding someone for the coding :)

andig, 2015-02-02:
> It supports touch devices, zooming, range selection and much more.

@Xento: Apart from the licensing issues: what *exactly* are you missing with flot? As far as I understand we have touch support, zooming, we have range selection and we certainly have "much more".

Instead of describing the solution please start by explaining the feature you're missing.

PS.: Switching to highcharts would likely be 5~10 days effort exercise. If anybody wants this outside of the Volkszaehler core right now feel free to contact me.

andig, 2015-02-02:
> Volkszähler is not solely used in non-commercial environments so implementing it as a default could lead to involuntary license violations. 

By the respective users. Would be their responsibility if we wanted to play it that way.

> Also the GPLv3 used here doesn't tolerate an NC-part so I don't think it would even be possible to create a bundle containing both codes.

What is NC? Indeed, the CC license of HighCharts (http://creativecommons.org/licenses/by-nc/3.0/) doesn't seem to be compatible with GPLv3 (https://wiki.creativecommons.org/Frequently_Asked_Questions).

Xento, 2015-02-02:
Ok, here is what i'm missing. 
Mostly i use Volkszähler on my iphone. So i want to Zoom in and out with two Fingers and scroll through the Timeline with one Finger.

andig, 2015-02-02:
> So i want to Zoom in and out with two Fingers 

There is a pinch plugin for flot, I'll need to take a look.

> and scroll through the Timeline with one Finger. 

Pan is supported as well.

See http://justindarc.github.io/flot.touch/

Xento, 2015-02-02:
I added the touch plugin for flot, but it is very ugly.
It seems that you can only zoom the graph like an image. When you zoom with highstock it generates a new graph with the selected timespan.

andig, 2015-02-03:
> I added the touch plugin for flot, but it is very ugly.

So? The plugin is 4 years old and needs a little work. That is what open source is about- build on what's there and start improving it.

andig, 2015-02-10:
I've pushed an update to the plugins site. It now has what we need. All that's missing is the frontend integration to dynamically load the data.

closed 2015-02-10 by andig