volkszaehler/volkszaehler.org #240: Error from mbus.pm on raspberry pi

kami83, 2015-02-11:
Hi,

i am connecting to an Mbus Meter and i always get this strange error:

 ./mbus-cmd --sqlout
Argument "ddebbd" isn't numeric in multiplication (*) at /usr/local/lib/site_perl/mbus.pm line 512.
INSERT INTO zaehler SET pos="0",vif="12",dif="6",value="00003622",unit="kWh",zweck="Heizenergie",serial="39202522",valArt="normaler Wert",count="0";
INSERT INTO zaehler SET pos="1",vif="12",dif="19",value="78.899",unit="m^3",zweck="Volumen",serial="39202522",valArt="normaler Wert",count="0";
INSERT INTO zaehler SET pos="2",vif="59",dif="59",value="0",unit="m^3/h",zweck="Fliessgeschwindigkeit",serial="39202522",valArt="Fehlerwert",count="0";
INSERT INTO zaehler SET pos="3",vif="10",dif="90",value="22.9",unit="C",zweck="Vorlauftemperatur",serial="39202522",valArt="normaler Wert",count="0";
INSERT INTO zaehler SET pos="4",vif="10",dif="94",value="22.9",unit="C",zweck="Ruecklauftemperatur",serial="39202522",valArt="normaler Wert",count="0";
INSERT INTO zaehler SET pos="5",vif="140 16",dif="6",value="00000000",unit="kWh",zweck="Heizenergie",serial="39202522",valArt="normaler Wert",count="0";
INSERT INTO zaehler SET pos="6",vif="10",dif="166 24",value="0000",unit="h",zweck="Betriebszeit",serial="39202522",valArt="normaler Wert",count="0";
INSERT INTO zaehler SET pos="7",vif="129 32",dif="253 26",value="0",unit="",zweck="Fehler (vife=26)",serial="39202522",valArt="normaler Wert",count="0";

OR:

./mbus-cmd --sqlrawout
INSERT INTO rawdata SET serial="39202522",data="\104\57\57\104\8\3\114\34\37\32\57\36\35\47\4\126\112\0\0\12\6\34\54\0\0\12\19\153\136\7\0\59\59\189\235\221\10\90\40\2\10\94\40\2\140\16\6\0\0\0\0\10\166\24\0\0\129\32\253\26\0\115\22";
Argument "ddebbd" isn't numeric in multiplication (*) at /usr/local/lib/site_perl/mbus.pm line 512.

You can see where is the Problem with "ddebbd" ??

Thanks a lot for help.

Cu kami

andig, 2015-02-19:
Why do you want to output any SQL? What are you trying to achieve?

tabacha, 2015-02-19:
Hi kami,
I will have a look at this.

tabacha, 2015-02-19:
Hi Kami,
is the error message "Argument "ddebbd" isn't numeric in multiplication (*) at /usr/local/lib/site_perl/mbus.pm line 512." only displayed in the second command?

I do not know if this code is ever run on a non intel architecture. Perhabs there is a byte order problem.

Is there a serial number writen on the connected meter? Is it correct? 
How many Meters are connected.

Do you have other mbus software? If yes, what values are displayed there?

Are the values 3622 kWh Heating and
78899  m³ volume flow correct? 

Can you please run:
```bash
mbus-cmd --debug --sqlout --sqlrawout >log_issue_240-0.log 2> log_issue_240-1.log
mbus-cmd --debug --sqlout --sqlrawout >log_issue_240-3.log 2>&1

```
Put the log files in the issue tracker.

Best Regards
Sven

kami83, 2015-02-20:
Hi,
i am sorry i cant post you any error. The error is gone. I am now using 3 meters with the PW3 and everything works fine.

Sorry.

Cu kami

tabacha, 2015-02-20:
Hi Kami,

good news.  No problem.

Sven
@andig magst du den Bug bitte schließen?

closed 2015-02-20 by andig