volkszaehler/volkszaehler.org #242: /misc/controller/1wire obsolete

andig, 2015-02-19:
Is replaced by 1wirevz or not? 

Suggesting to delete.

r00t-, 2015-02-19:
links are overrated, yes?
https://github.com/volkszaehler/volkszaehler.org/tree/master/misc/controller/1wire

r00t-, 2015-02-19:
from what i can tell, this is different stuff.
1wirevz uses sysfs and a kernel driver on raspberry pi to read sensors.
the stuff in 1wirevz is
- a wrapper for 'digitemp' (userspace tool that speaks 1wire by abusing an uart/rs232 port)
- a script to get 1wire data from ethersex via ecmd
so not replaced by 1wirevz

r00t-, 2015-02-19:
i think this is more about:
do we want to accept/keep stuff contributed by third parties?
and how much maintainance activity do we want to expect from the authors?
same as the discussion that ensued here: http://volkszaehler.org/pipermail/volkszaehler-dev/2015-February/004257.html
i suggest you open a separate issue for that.

my personal opinion would be that making that much demands discourages contribution.
while i would suggest to keep trivial scripts on the wiki. (which the 1wire scripts might aswell be)

andig, 2015-02-20:
> do we want to accept/keep stuff contributed by external authors into the repository?

opened https://github.com/volkszaehler/volkszaehler.org/issues/246

> my personal opinion would be that making that much demands discourages contribution.

we should find the right place.

+1 for move to wiki

justinotherguy, 2015-03-01:
a note on 1wire:
this is useful for collecting values off of 1wire devices attached to atmega boards running ethersex.
Although this is not "main stream" anymore nowadays, these setups are still useful (think: power consumption).
Having them in the wiki is not as comfortable as git - seeing their last commit date 3 years ago, I'm fine with that :-)
Question is: what is the criteria for "git or wiki?" the number of lines of code?

andig, 2015-03-02:
I'll add the results from this issue to the README and close

PR #260 (2015-03-02): Misc
Close #242 and #243
closed 2015-03-02 by andig