volkszaehler/volkszaehler.org #243: /misc/controller/current_cost obsolete

andig, 2015-02-19:
5 years old, highly specific, website link dead.

Suggesting to delete.

r00t-, 2015-02-19:
https://github.com/volkszaehler/volkszaehler.org/tree/master/misc/controller/current_cost

the documentation could be better, but it probably related to this:
http://www.currentcost.com/products.html

note that the author is active on our mailing list, so how about you ask him?

see also my comment here: https://github.com/volkszaehler/volkszaehler.org/issues/242#issuecomment-75157977

andig, 2015-02-20:
current_cost has nothing todo with vz but maintains its own rrd db. As it is it's off-topic.

r00t-, 2015-02-21:
Date: Sat, 21 Feb 2015 02:02:28 +0100 
From: Roland Kletzing <devzero@web.de> 

Hi, 

also ich hab kein Problem damit wenn ihr das aus dem Volkszähler Projekt rausschmeisst. 
Steffen wollte wohl mal Currentcost-Support in den vzlogger einbauen und hatte das nur als Basis/Beispiel genommen....

Auf umweltsuen.de war auch nur nen redirect aufumweltsuende.wordpress.com, was mein in letzter zeit reichlich vernachlässigter blog ist....

schmeiss es raus - der code ist ja noch über meine website verlinkt - und interessiert auch eh kaum einen. currentcost ist in D. auch nicht wirklich angesagt....

Grüsse
Roland

andig, 2015-03-02:
> schmeiss es raus - der code ist ja noch über meine website verlinkt - und interessiert auch eh kaum einen. currentcost ist in D. auch nicht wirklich angesagt....

Würde ich gemäß Kommentar löschen?

andig, 2015-03-02:
Closed via #260

closed 2015-03-02 by andig