volkszaehler/volkszaehler.org #246: RFC: Criteria for repo contributions

andig, 2015-02-20:
To keep the repo in a current state I'd suggest the following ruleset for contributions:

  - core repo (i.e. vz.org and vzlogger) **must** address a broader public (> 1 user)
  - contributions (i.e. anything in the repo) **must have** at least one maintainer
  - maintainers **must** subscribe the dev mailing list
  - maintainers **should** subscribe the github repo

Obsoleting contributions:

  - outdated code (i.e. that is not working anymore/ with current vz, os etc) should either be fixed or removed
  - additional to removal move to wiki is optional

Comments welcome.

jahir, 2015-02-20:
sounds reasonable to me

mbehr1, 2015-02-20:
Fine for me as well.

I’d suggest to keep a file „MAINTAINERS“ (like kernel.org <http://kernel.org/>) that we get a clear picture who is maintaining what.

> To keep the repo in a current state I'd suggest the following ruleset for contributions:
>
> core repo (i.e. vz.org and vzlogger) must address a broader public (> 1 user)
> contributions (i.e. anything [...]

Gruß
Matthias

andig, 2015-02-20:
One additional suggestion:

  - contributions **must** use the volkszaehler API

This one would e.g. hit 1wire

> I’d suggest to keep a file „MAINTAINERS“ 

In principle yes, but it's a 4 persons list, not even worth writing down:

  - vz.org: jahir + me
  - vzlogger: mbehr + root

r00t-, 2015-02-20:
copying my statement from https://github.com/volkszaehler/volkszaehler.org/issues/242#issuecomment-75157977

i think this is more about:
do we want to accept/keep stuff contributed by external authors into the repository?
and how much maintainance activity do we want to expect from the authors?
same as the discussion that ensued here: http://volkszaehler.org/pipermail/volkszaehler-dev/2015-February/004257.html
i suggest you open a separate issue for that.

my personal opinion would be that making that much demands discourages contribution.
while i would suggest to keep trivial scripts on the wiki. (which the 1wire scripts might aswell be)
maybe we should create a contrib dir in the repository, to make it clearer that we are not directly responsible for the stuff?

r00t-, 2015-02-20:
i don't agree with that kind of policy.
ESPECIALLY as your next step seems to be to apply the same policy to wiki entries.

imho volkszaehler.org is not just about it's specific software,
but also a collection of know-how and code for smart metering in general.
that's also how i got involved in the first place.

as such, we should be more open to contributions than what is suggested here.

i would agree that stuff not related to the middleware/api at all should probably not be
in the "volkszaehler.org" repository.
maybe we could create an extra repo for hosting such stuff.

waiting for a statement from @justinotherguy...

andig, 2015-02-20:
> ESPECIALLY as your next step seems to be to apply the same policy to wiki entries.

if you want to know what my "next steps" as you call them are feel free to ask me (as you're regularly suggesting) instead of speculating

r00t-, 2015-02-20:
i am refering to https://github.com/volkszaehler/vzlogger/issues/103

justinotherguy, 2015-03-01:
I'm also fine with the policy. I'd replace "removed" with "marked as obsolete", though.
Objective:
- make sure (new) users only see code, that is actively maintained (avoid people using "the wrong code" by accident)
- provide some old/non working/incompatible/... code to users explicitely looking for it (avoid complete extinction of code that has been contributed while ate the same time making sure we don't maintain a dump)
I can imagine cases where we get rid of most references, if the code has been replaced by something almost identical like vzlogger-c.

andig, 2016-06-25:
VZ meeting: not a current issue due to small team size

closed 2016-06-25 by andig