volkszaehler/volkszaehler.org #268: Falsche Berechnung des Verbrauchs

milefork, 2015-03-08:
Hi,

ist es irgendwie möglich nur die positiven Werte auf den Verbrauch zu addieren? Also ich lese die 16.7.0 vom Stromzähler und mich würde es interessieren wie viel ich wirklich aus dem Netz gezogen habe. Aber bei der zusammen Rechnung des Verbrauches scheinen die negativen Werte auch aufaddiert zu werden. Hier mal ein paar Fotos:

http://puu.sh/gsnOC/337f25a1f2.png

Das nächste Bild kommt vom roten Bereich:

http://puu.sh/gsnTX/fd829fd2f1.png

Und mal eins wenn die Anlage Brummt, wobei man sieht das der Verbrauch definitiv im Minus ist, aber ich auch definitiv was aus dem Netz gezogen habe:

http://puu.sh/gso2Z/71fd2995fd.png

Kann mir jemand nen Tipp geben oder ist das nen Bug?

andig, 2015-03-09:
Moin Christian,

Wenn ich Dich richtig verstehe möchtest Du den Teil der Kurve unter de x-Achse bei der Verbrauchsberechnung gerne ignorieren? Das ist leider nicht vorgesehen.

Viele Grüße, Andreas

milefork, 2015-03-09:
Okay Danke für die schnelle Antwort!
Kann die middleware das irgendwie errechnen oder muss ich mir selbst nen Script schreiben?

andig, 2015-03-09:
> Okay Danke für die schnelle Antwort!
> Kann die middleware das irgendwie errechnen oder muss ich mir selbst nen
> Script schreiben?

Skript geht immer, Middleware leider nein. Hat Dein Zähler nicht evtl.
getrennte OBIS Codes für Einspeisung und Bezug die Du loggen kannst?

Viele Grüße,
Andreas

jahir, 2015-03-09:
Hallo,

wie Andreas schon vorgeschlagen hat, solltest du möglichst einen anderen Kanal dafür benutzen. Den Momentanwert der Leistung zur Ermittlung der Energie zu benutzen ist theoretisch natürlich korrekt, aber wenn man das nicht sauber macht doch recht fehlerbehaftet.
Dein Zähler müßte doch auch Bezug und Einspeisung (als Zählerstand) getrennt ausgeben. damit bekommst du die genauen Zahlen und kannst auch den Eigenverbrauch berechnen.

milefork, 2015-03-10:
Ja den habe ich natürlich auch. Aber ich glaube ich habe ihn falsch Konfiguriert, im Moment läuft er so:

http://puu.sh/gtQc4/c4ecb43da0.png

Kann es sein das ich ihn nicht als Leistungswert hätte konfigurieren sollen? Aber ich meine ich hatte diesen auch mal als Zählerstand, aber da habe ich die Auflösung nicht hinbekommen.

andig, 2015-03-10:
Meist helfen vollständige Informationen statt rätselraten:

- was für ein Zähler?
- wie angeschlossen?
- welcher Kanaltyp?
- welcher OBIS Code"

Viele Grüße, Andreas

jahir, 2015-03-10:
"Elektrische Energie (Leistungswerte)" ist Kanaltyp "power", das ist natürlich falsch.
Mit "electric meter" müßte die Anzeige dann stimmen. (Die Rumgezappel bei mir kommt von der schlechten Zählerauflösung und dem langsamen Auslesen.)
[image]

jahir, 2015-03-10:
Wobei mir gerade auffällt, daß die "574W" hier ja quatsch sind. Die Leistung liegt ab 17:55 bei ca. 7W (rechnerisch). Die 574W kommen vom nächsten Morgen, die zweite Zählerstandsänderung:

[image]

In den Werten sieht das dann so aus:

2015-03-09 16:58:16;1727.232;1
2015-03-09 17:02:11;1530.339;1
2015-03-09 17:07:33;1120.483;1
2015-03-09 17:19:18;510.646;1
2015-03-09 17:55:07;167.454;1
2015-03-10 07:36:13;7.307;1
2015-03-10 07:46:40;574.178;1

bzw.

2015-03-09 17:02:11;29030.5;1
2015-03-09 17:07:33;29030.6;1
2015-03-09 17:19:18;29030.7;1
2015-03-09 17:55:07;29030.8;1
2015-03-10 07:36:13;29030.9;1
2015-03-10 07:46:40;29031;1

Der letzte Wert sollte wohl eher nicht ausgegeben werden, der wird ja auch nicht mehr angezeigt. (Der vorletzte auch nicht, aber den brauchen wir halt, um einen "aktuellen" Wert anzuzeigen)
Naja, anderes Problem.

milefork, 2015-03-10:
@andig also das hier ist mein Zähler: http://puu.sh/gupA2/c836ed2492.jpg 

Obis Codes die ich auslese:

1.8.0 Bezug Netz
2.8.0 Einspeisung Netz
16.7.0 Momentanleistung

andig, 2015-03-10:
Ich steh auf dem Schlauch.

> Hab testweise mal die 1.8.0 auf electric meter gemapped 

Wieso? "1.8.0 Bezug Netz" dürfte eine Leistung sein, also Power.

> Also ich lese die 16.7.0 vom Stromzähler und mich würde es interessieren wie viel ich wirklich aus dem Netz gezogen habe.

Das wäre ja dann die "1.8.0 Bezug Netz", also einfach die richtig konfigurieren und es passt.

PS.: Das ist sicher kein Bug und gehört damit eigentlich auf die Mailingliste...

milefork, 2015-03-10:
Okay weiß ich fürs nächste mal bescheid, wie erreiche ich die Mailing Liste?

closed 2015-03-10 by milefork

jahir, 2015-03-10:
1.8.0 ist ein Zählerstand ("Wirkarbeit Bezug")

milefork, 2015-03-10:
@jahir also so erstellen? http://puu.sh/guthK/70cb405eb2.png

andig, 2015-03-10:
Sollte passen. 

Viele Grüße,
Andreas

milefork, 2015-03-10:
Okay muss das dann so aussehen? http://puu.sh/guxo2/52763213c6.png 
Jetzt muss ich nur noch die Auflösung anpassen oder?

andig, 2015-03-10:
Ja, das sieht doch nicht schlecht aus. Du musst aber nicht jeden Schritt abnehmen lassen, einfach machen ;) Auflösung ist vmtl 1000 wenn Dein Zähler kWh ausgibt?

Viele Grüße, Andreas

milefork, 2015-03-10:
Jo ist 1000 hab schon alles fertig, danke nochmal für eure Hilfe auch wenn's hier eigentlich nicht hingehörte ;)

andig, 2015-03-10:
Immer gerne!