volkszaehler/volkszaehler.org #270: Deleting channel raises constraint exception

mmame, 2015-03-10:
When trying to delete a channel, a constraint exception is thrown.

How to reproduce:
1. Create a S0 channel (e.g. "Gaszaehler")
2. Put some data in
3. Try to delete channel over frontend ("i" icon, "Löschen") ->

../middleware.php/entity/XXXXX.json?operation=delete:

An exception occurred while executing 'DELETE FROM entities WHERE id = ?' with params [10]: SQLSTATE[23000]: Integrity constraint violation: 1451 Cannot delete or update a parent row: a foreign key constraint fails (`volkszaehler`.`aggregate`, CONSTRAINT `FK_B77949FF72F5A1AA` FOREIGN KEY (`channel_id`) REFERENCES `entities` (`id`))

andig, 2015-03-11:
How current is your code? What's the output of `git log`?

mmame, 2015-03-11:
What I did: 
1. downloaded vz image from here: https://demo.volkszaehler.org/downloads/volkszaehler_image.rar
2. Did a "git clone" yesterday on volkszaehler

So I expect to be up to date.

I'll provide the git log as soon as I am back home :-)

andig, 2015-03-11:
apart from the `git log` please also try to set `devmode=true` in the config file and retry. This will effectively clean the caches. Does it still error?

andig, 2015-03-14:
ping

mmame, 2015-03-14:
Sorry for the delay. My guess is there is an issue within prebuilt image 2015_11_02 - not in the middleware itself. Unfortunately, I killed the whole installation by trying to fix the issue :-> Anyway, I'll restart with a clean installation according http://wiki.volkszaehler.org/software/middleware/installation
(not the image I used last time). If there are any reproduceable issues I'll come back to you.

closed 2015-03-14 by mmame