volkszaehler/volkszaehler.org #280: Apache httpd 2.4 changes in access configuration

jahir, 2015-03-26:
(This is just a reminder for the near future)

The access config was changed in httpd 2.4. We will need "Require" instead of "Allow/Deny from". See http://httpd.apache.org/docs/current/mod/mod_authz_host.html for more details.

We use this in two files:
* .htaccess
Replace the "Deny/Allow ..." with "Require local".
Though I don't see a reason why one should want to access the vz root dir via http, so "Require all denied" is probably the better choice.

* htdocs/.htaccess
"Require all granted" instead of "Allow from all".

Unfortunately, I couldn't find a way to have both in the same file and enable to appropriate one with some <If*> keyword (the module name is still mod_authz_host and there's no way to check the version), so we'd have switch at some point (and keep the old directives as comments), maybe after debian jessie is stable.

adlerweb, 2015-03-26:
Wordpress is using mod_authz_core.c to detect 2.2 vs. 2.4 - maybe worth a try...
https://plugins.trac.wordpress.org/changeset/1028901/akismet/trunk

jahir, 2015-03-31:
@adlerweb kind of dirty, but I guess that's the best we have. and if it's good enough for the wordpress people, I guess it's ok for us too

PR #283 (2015-04-01): Apache 2.4 compatibility
Fix #280
closed 2015-04-01 by jahir