volkszaehler/volkszaehler.org #287: Wrong composer.phar

Spanni26, 2015-04-07:
When calling "bash install.sh" it only checks if $COMPOSER is not empty. 

echo "checking composer..."
COMPOSER="$vz_dir/composer.phar"
if [ ! -n "$COMPOSER" ]; then
                for f in composer composer.phar; do
                               COMPOSER=$(which $f 2>/dev/null || true)
                                test -n "$COMPOSER" && break
                 done
fi

In my opinion it should be if [ ! -f "$COMPOSER" ]; then....so it checks for composer if $vz_dir/composer.phar is not found......

At the moment it fails cause there is no composer.phar in $vz_dir

jahir, 2015-04-07:
Yes, the install script is broken in more than one way. I rewrote some parts of it (including the part you mention) but failed to open a pull request due to some usage git usage issues on my side. Will try to catch up asap.

andig, 2015-04-07:
Should be fixed, please retest: https://github.com/volkszaehler/volkszaehler.org/commit/6ec5b312b9be305fb7d38551b2ecaced9e1651a3

jahir, 2015-04-07:
reworked install script in https://github.com/volkszaehler/volkszaehler.org/pull/288

(btw, I merged your fix from master into development so I won't get merge errors. We should not let diverge the two branches too much...)

andig, 2015-04-08:
@jahir right now dev is more for validating ideas- the commits on master are not identical. It should suffice to rebase dev on master from time to time.

andig, 2015-04-15:
Fixes by #288

closed 2015-04-15 by andig