volkszaehler/volkszaehler.org #302: Mix/max values change when zooming in depending on resolution

andig, 2015-05-15:
We should think about always calculating/displaying "real", highest-resolution min/max values. This would affect aggregation. We should rethink removing custom aggregation code and using standard `interpreter` functions instead now that they are based on `iterators` and don't cause memory issues anymore.

gitaeuber, 2015-12-10:
i strongly support this issue

the chart looks good to me as is.
i don't know how it would look like with other aggregation algorithms, but it would be nice if the min and the max values are shown in the chart
this way a user can see where to zoom in to have a closer look

thanks

andig, 2015-12-11:
>the chart looks good to me as is.

So no changes to the chart?

> i don't know how it would look like with other aggregation algorithms, but it would be nice if the min and the max values are shown in the chart

...or do you propose to change the chart? How should it look like? Show a min, a max and a medium line? Lots of lines if you have more than one uuid.

gitaeuber, 2015-12-11:
I'd just leave the chart as is but add the peaks of (all) min or max values like this:

http://peltiertech.com/Excel/pix2/MarkMarkers03a.gif

(or if more than one point in time have the same max value:
http://peltiertech.com/Excel/pix2/MarkMarkers05.gif )

from: http://peltiertech.com/Excel/Charts/FormatMinMax.html

So the aggregation should not cut off these values from the chart.

thanks

andig, 2015-12-11:
> I'd just leave the chart as is but add the peaks of (all) min or max values like this:

I can only repeat myself. If the chart shows "12" the min/max values may be 10 and 17. Also the chart does not easily support adding arrows. This is not the way to go.

andig, 2015-12-27:
> Or something like this (ugly even if correct):

> I'd just leave the chart as is but add the peaks of (all) min or max values like this:

None of this does help. You can have min/max values for every single interval in the chart. Unless we get a clear model _how_ this should look there won't be any fixes.

closed 2016-06-25 by andig