volkszaehler/volkszaehler.org #313: website: css 'bottom' class breaks layout on firefox

andig, 2015-06-03:
[image]

Fix: remove `bottom` class.

justinotherguy, 2015-06-05:
Hm - I just gave it a look (Win8.1, FF38.0.1; Win8.1, FF 38.0.5; OSX 10.10.3, FF 38.0.1); it looks fine on these; could you try again with an empty profile?

andig, 2015-06-05:
FF38.0.5 Win7x64. Es sind zwei Dinge komisch:

  - Aus irgendeinem Grund funktioniert die absolute Positionierung von `.bottom` nicht. Die Elemente scheinen nicht unten an der Box zu kleben sondern am vorhergehenden Text. Trotzdem überlappen sie sich. Bei Chrome passt es auch bei mir. 
  - Bei mir ist die mittlere Box schmaler und hat 6 Zeilen Text (links nur 4). Damit kommen sich Text und absolute Positionierung in die Quere. Evtl. ein Thema von Default Fonts, Schriftgrößen etc.

Du kannst das nachvollziehen wenn Du z.B. die Breite der mittleren Box (`.container12 .grid_3`) von 220px auf 200px reduzierst. Was ich nicht verstehe ist wie die absolute Positionierung überhaupt funktioniert. Anders gefragt: welches CSS sorgt in umgebender Box/ vorhergehendem Text dafür dass unten ausreichend Platz für den Link bleibt? 

Die absolute Positionierung außerhalb des Textflows scheint nicht 100%ig stabil. Brauchen wir die denn oder würde es nicht reichen den Link am Ende des Absatzes in eigener Zeile erscheinen zu lassen?

andig, 2015-06-12:
> Aus irgendeinem Grund funktioniert die absolute Positionierung von .bottom nicht

Es ist ganz klar. `.bottom` bezieht sich auf `div.box` und die hat nicht 100% Höhe wenn nicht alle Spalten gleiche Anzahl Zeilen haben.

Daher nochmal der Vorschlag: auf `.bottom` verzichten und die Links einfach in neue Zeile setzen.

andig, 2015-07-07:
Ping @justinotherguy

UdoSchake, 2015-08-02:
[image]

Ich sehe das Problem nicht. FF 39.0

Gruß
Udo

andig, 2016-03-21:
Ich geb auf :(

closed 2016-03-21 by andig