volkszaehler/volkszaehler.org #318: Water meter does not show correct actual

bgewehr, 2015-06-13:
When using a water meter it shows the last or whatever value in the actual field and as the actual value when using the timeline feature of the graph - even when actually no water is flowing through the meter and the graph is at zero level.

[image]

This behaviour was generated by the following settings:
[image]

Shouldn't the actual value be 0 if no water flows through the meter?

andig, 2015-06-14:
Where should a value of zero come from? If s0vz doesnt receive pulses it can't push them to the middleware. The middleware in turn cannot know if no pulses means no data yet or zero.

andig, 2015-06-14:
To fix you would need to somehow write 0 whenever no data received for a certain amount of time. I beleive we have this capabilty in vzlogger meanwhile.

mbehr1, 2015-06-14:
Didnt we discuss once whether we should change s0 handling to "impulses per time unit" instead of one data set er impulse?
(Btw does your sensor really return up to 60-70 impulses per s?)

Gruß
Matthias

Sent from a mobile device.

bgewehr, 2015-06-14:
In 99,99% of all time it means no flow when no pulses are registered. Only in case of an error no pulses can possibly NOT mean no flow. So why do we make 99,99% false positive flow detection only to avoid the error in 0,01% of time?

bgewehr, 2015-06-14:
Yes, the cheap flow meters detect 600l/h with 82Hz.

bgewehr, 2015-06-14:
i think we should configure a falling period in which no received pulses mean falling to zero. Be it an hour you get a slow reacting curve. Be it 1s it will react immediately to zero if no more pulses are seen.

andig, 2015-06-14:
> Didnt we discuss once whether we should change s0 handling to "impulses per time unit" instead of one data set er impulse?

@mbehr1 I remember we did. What is `aggtime` doing when we no reading is received? Would it send 0 to the middleware?

@bgewehr this is only supported in vzlogger, so you'll have to wait until https://github.com/volkszaehler/vzlogger/pull/169 is merged anyway.

> In 99,99% of all time it means no flow when no pulses are registered. 

Maybe, but it needs be configurable. My Ferraris may only trigger an impulse every few minutes and still have actual consumption for base load. 

> i think we should configure a falling period in which no received pulses mean falling to zero. Be it an hour you get a slow reacting curve. Be it 1s it will react immediately to zero if no more pulses are seen.

Right. Lets see waht @mbehr1 says about `aggtime`. I don't think though that it would make sense to implment this in `s0vz` at this stage.

fostex68, 2015-06-14:
Ich finde das ein super Idee. Bei mir erfasst ein netIO die S0-Pulse, der sendet 
nichts an die Middleware wenn keine Impulse kommen (Ich weiß auch nicht wie man 
das ändern könnte, ich steige bei dem Code für watchasync nicht durch). Das 
Problem habe ich schon lange mit S0-Stromzählern, dass die Kurve nicht vorhanden 
ist aber der aktuelle Wert nicht 0 ist. Seit ein paar Wochen ist die Heizung aus 
aber der aktuelle Wert ist 54W...
Ein einstellbarer Timeout in der Middleware würde das elegant und einfach auch 
für ältere Erfassungshardware lösen.

r00t-, 2015-06-14:
actual <-> tatsaechlich
current <-> aktuell

http://de.wikipedia.org/wiki/Falscher_Freund

bgewehr, 2015-06-14:
Right. We need the actual current value, that's what it's all about!

bgewehr, 2015-06-15:
I have put curl -X POST -d "value=0" http://<vz-ip>/middleware.php/<unid>.json in a cronjob every minute and my current value falls down to zero when the water valve has closed after at least a minute. Not pretty but a first workaround.

andig, 2015-06-15:
>> Didnt we discuss once whether we should change s0 handling to "impulses per time unit" instead of one data set er impulse?

> I remember we did. What is aggtime doing when we no reading is received? Would it send 0 to the middleware?

I think we've discussed this when talking about handling 'duplicates'. We would need something similar here: write a value to the middleware even if not reading received from meter. Drawback: if the real reading appears shortly after, frontend will show an ugly peak. If we want to make it complicated we'd produce a "conditional" zero reading that time x that is only being sent if no other reading appears within time x afterwards.

What I don't like is the proliferation of config settings. Would it make sense to remove 'duplicates' and replace it with something like a 'maxreadinginterval'?

jahir, 2015-06-15:
@bgewehr That's what I also do (in a slightly different use case), but pushing fake values into the database is kind of hackish.
I guess it should not be a big issue to add a "assume_zero_after" or "min_output" (l/h, kWh or whatever) property and let the middleware insert a 0 value in the output for the frontend (but with which timestamp?) OTOH, wouldn't it be enough to just add a fake 0 at the end of the series if the last impulse was more than x seconds before the interval end? For in-between-gaps you would get a very low usage (which is technically correct), which good enough IMO.

andig, 2015-06-15:
> OTOH, wouldn't it be enough to just add a fake 0 at the end of the series if the last impulse was more than x seconds before the interval end? For in-between-gaps you would get a very low usage (which is technically correct), which good enough IMO.

I guess it's the same basic problem as during the 'duplicates' discussion (https://github.com/volkszaehler/vzlogger/issues/98). Es kommt am Ende darauf an um welche Art physischer Größe es sich handelt und was man bei der Messung erreichen will. Geht es z.B. Darum Peaks zu finden so passt der Ansatz nur eine letzte Null zu schreiben nicht. 
Und ganz wichtig: das Frontend fragt je nach Zoomstufe Aggregate ab. Es gibt also keine Garantie das eine einzelne Null auch tatsächlich gelesen wird- genau so gut kann diese auch in einem Paket mit einem späteren Folgewert landen. Je mehr Nullen es zwischendurch gibt desto höher die Wahrscheinlichkeit (nicht Garantie) auch einen Ausgabewert von Null zu erhalten.

andig, 2015-06-25:
> I guess it should not be a big issue to add a "assume_zero_after" or "min_output" (l/h, kWh or whatever) property and let the middleware insert a 0 value in the output for the frontend (but with which timestamp?) 

I don't like it. Right now all "truth" about meter data is in the database. We would start distribution that logic across components, too.

Also- what would make such a setting better than handling it in vzlogger?

Btw: "could we/should we" discussions are better suited for the mailing list for broader participation.

Should we close this issue?

bgewehr, 2015-06-25:
If this is continued in the vzlogger issues, closing here is OK for me...

andig, 2015-06-27:
> If this is continued in the vzlogger issues, closing here is OK for me...

For specific vzlogger issue please open at vzlogger. For general discussion how to treat missing values please use the mailing list to get broader feedback.

closed 2015-06-27 by andig