volkszaehler/volkszaehler.org #338: Zusätzliche Option für Kanaltyp (Ventil)

bernemann, 2015-07-28:
Hi,

wenn ich nur Statusänderungen und minütliche Werte logge (Kanaltyp Ventil) springt der Graph schon beim vorherigen Messpunkt auf den neuen Messwert. In meinem Anwendungsfall ist er somit nach links verschoben.

Beispiel 4 Messpunkte:

17	1438024200000	0	2015-07-27 21:10:00.0000
17	1438024140000	1	2015-07-27 21:09:00.0000
17	1438024080000	1	2015-07-27 21:08:00.0000
17	1438024020000	0	2015-07-27 21:07:00.0000

erzeugt den Graph

[image]

Gewünscht hätte ich mir hier: Pumpe läuft von 21:08:00 Uhr - 21:10:00 Uhr. Eine Option hierfür in der Kanaldefinition wäre sehr nützlich.

andig, 2015-07-28:
Display Style: Steps sollte Konfiguration von step_before und step_after ermöglichen (analog rickshaw/d3js).

andig, 2015-10-11:
In welchem Intervall misst Du? Ständig oder minütlich?

bernemann, 2015-10-12:
Derzeit messe und logge ich alle ankommenden Nachrichten, der Bus sendet "zufällig" alle 20-40 Sekunden etwas. Das minutliche Log oben ist nur zur Veranschaulichung gedacht.

closed 2016-07-29 by andig