volkszaehler/volkszaehler.org #343: FF on OSX shows "Javascript Runtime Error SecurityError: The operation is insecure."; does not seem to affect functionality

justinotherguy, 2015-09-09:
Hi,
after the update to the current git head I get a popup saying:
"Javascript Runtime Error 
SecurityError: The operation is insecure."
After confirming the dialog I can't seem to notice any impact on functionality, though:
the graphs are rendered, the stats are shown, I can hover the graphs and the values are shown, I can subscribe to new channels.
I found this bugs that could be related: https://stackoverflow.com/questions/21642277/security-error-the-operation-is-insecure-in-firefox-document-stylesheets
The popup does not show on Safari and Chrome on the same machine.
FF is v 40.0.3 on OSX.
Keep up the great work, guys! :)
Regards, J.

andig, 2015-09-09:
Any line number/ context for the error?

justinotherguy, 2015-09-09:
yes, console says: 
autobahn.min.js:62:0

does that help?

justinotherguy, 2015-09-09:
update:
only happens, when I access https://demo.volkszaehler.org; doing so in Safari does not cause a problem.
Chrome says:
"Javascript Runtime Error
Uncaught SecurityError: Failed to construct 'WebSocket': An insecure WebSocket connection may not be initiated from a page loaded over HTTPS."
-> something with the new web sockets?

andig, 2015-09-10:
Please retest after git pull'ing. This should also take are of Stefan's error from the ML.

closed 2015-09-10 by andig

justinotherguy, 2015-09-10:
yes, confirmed - fixed!
Thanks a lot - keep up the excellent work, Andreas! :)