volkszaehler/volkszaehler.org #348: Deletion of data should be supported

andig, 2015-09-24:
- single timestamp
  - from .. to range
  - single or multiple channels

bgewehr, 2015-09-24:
My experience is:
Delete the channel and all data of the channel is deleted. Am I wrong?

With phpmyadmin I am able to just type delete * from data where ID = 17 and timestamp is between 1234567 and 2345678

(Syntax not exact)

Isn't that enough?

andig, 2015-09-24:
On Thu, Sep 24, 2015 at 3:29 PM, Bernd Gewehr <notifications@github.com>
wrote:

> My experience is:
> Delete the channel and all data of the channel is deleted. Am I wrong?
>
> With phpmyadmin I am able to just type delete * from data where ID = 17
> and timestamp is between 1234567 and 2345678
>
> (Syntax not exact)
>
> Isn't that enough?
>

It is, but it's not an api. Worse: the api doesn't complain when delete is
called but actually does nothing.

closed 2015-09-28 by andig