volkszaehler/volkszaehler.org #353: Nach Update json Fehler im Frontend

milefork, 2015-10-01:
Moin,

habe heute morgen mal mein Frontend geupdated: git pull; composer self-update; composer update; composer install

Laut meinen Access und Error logs geht auch alles wieder nur läd mein Frontend nicht mehr, dort ist ein javascript fehler: http://puu.sh/kuydl/abe25fb686.png es scheint so als ob json gar nicht existiert.

Kann mir da jemand helfen? Gerne auch über icq, jabber, irc (damit das hier nicht so viel wird)

Gruß

milefork, 2015-10-01:
Ah habs, ein dpkg-reconfigure php5-json hats gebracht....

closed 2015-10-01 by milefork

andig, 2015-10-01:
Interessant. Was für ein OS ist das? Gibts eine ordentlich exception von der Middleware wenn Du das nicht machst? Wäre schön zu verstehen ob die Fehlerbehandlung funktioniert!

milefork, 2015-10-01:
ja gab es! Schön ein einem roten Fenster, hab ich leider kein Screenshot mehr von aber man konnte dadurch auf Javascript schließen (stand was von Javascript).
Das OS ist ein Debian auf Stable Release Channel. Und der Fehler entstand wahrscheinlich dadurch, dass ich den Release Channel letzte Woche geändert hatte und php5-json dann nicht mehr richtig konfiguriert wurde.