volkszaehler/volkszaehler.org #357: JAVA POST an  middleware geht nicht mehr!

GeminiServer, 2015-10-01:
Hallo zusammen, 
mit der früheren Version vom middleware (ca. 1 Jahr Alt ) hatte es noch funktioniert. Nach Update geht es nicht mehr. Value wird sauber übergeben. Wenn ich mir die DATA vom UUID anschaue, wird immer der Default Wert: 1 anstatt der übergebenen Value geschrieben! Ich vermute die middleware schreibt die default 1, weil die middleware die übergebene Value nicht sauber annimmt. Ne Idee wo der wurm ist ;)

Hier ein JAVA Besipiel: (Compilieren und Kurz mal testen....)
```javascript
import java.io.BufferedReader;
import java.io.InputStreamReader;
import java.io.OutputStreamWriter;
import java.net.URL;
import java.net.URLConnection;

public class Main {

  public static void main(String[] args) throws Exception {
    URL url = new URL("http://localhost/sz/middleware.php/data/d59bd540-f67f-11e4-8edf-abb3094bd0b0.json");
    URLConnection conn = url.openConnection();
    conn.setDoOutput(true);
    OutputStreamWriter writer = new OutputStreamWriter(conn.getOutputStream());
    writer.write("value=97");
    writer.flush();
    String line;
    BufferedReader reader = new BufferedReader(new InputStreamReader(conn.getInputStream()));
    while ((line = reader.readLine()) != null) {
      System.out.println(line);
    }
    writer.close();
    reader.close();

  }
}
```

Danke & Gruß, 
GeminiServer

P.s.: mit curl gleiches verhalten!

```
curl --data "value=99" http://127.0.1.1/sz/middleware.php/data/d59bd540-f67f-11e4-8edf-abb3094bd0b0.json
```

andig, 2015-10-02:
Du machst einen POST Request? Häng die Parameter einfach an den GET an.

closed 2015-10-02 by andig

andig, 2015-10-02:
So ist es übrigens auch dokumentiert: http://wiki.volkszaehler.org/development/api/reference