volkszaehler/volkszaehler.org #363: Kanal erstellen: Überprüfung des Wertes “Auflösung“ auf Plausibilität

donhasso, 2015-11-13:
Hallo,

wird über Browser ein neuer Kanal erstellt und das Feld “Auflösung“ nicht ausgefüllt, so schließt zwar das Fenster beim Speichern, aber der neu angelegte Kanal erscheint nicht in der Auflistung unter dem Diagramm. Eine Fehlermeldung wäre sinnvoll.

andig, 2015-12-23:
Peng. Da waren gleich 2 Fehler drin:

  - die MW hat einige Fehler "verschluckt" und
  - das Frontend hat- solange HTTP 200 kam- Fehler komplett ignoriert

Beides ist behoben, muss aber sorgsam getestet werden da relativ grosser Umbau.

andig, 2015-12-27:
Ping. Wäre jetzt bereit zu Test.

andig, 2015-12-29:
ping @donhasso

donhasso, 2015-12-29:
Ääähhh, ich denke ich bin angesprochen. Was gibt es zu tun?

andig, 2015-12-29:
Du müsstest den PR testen. D.h. In Deine lokale Installation importieren und dann testen. Da recht umfangreich macht es i. Diesem Fall keinen Sinn die Änderungen von Hand zu übernehmen.

Klappt das mit Git + Google?

donhasso, 2015-12-30:
Hallo,

wenn ich es richtig verstanden habe, muss ich die 8 Files, welche Du geändert hast in meinem System löschen und die Neuen installieren?

Die Installation erfolgt über zum Beispiel:

sudo wget https://github.com/andig/volkszaehler.org/blob/831ca61df38478151672736185c7795b5388bf55/htdocs/frontend/javascripts/entities.js

andig, 2015-12-30:
Kannst Du so machen. Oder einfacher- wenn Deine Installation schon auf git basiert so:

    git checkout -b andig-vz.load-errors master
    git pull https://github.com/andig/volkszaehler.org.git vz.load-errors

Ausführen in /var/www/volkszaehler.org bei einer Standardinstallation.

Das hat den Vorteil dass Du per `git checkout master` sofort wieder auf den Hauptzweig zurück kommst.

donhasso, 2015-12-30:
Ok, bin nun wie folgt vorgegangen (bin ein DAU), konnte aber über das 
frontend keine Änderung sehen. Muss ich dazu noch etwas ändern oder soll 
ich ein Log mitlaufen lassen?

pi@raspberrypi /var/www/volkszaehler.org $ sudo git checkout -b 
andig-vz.load-errors master
M    htdocs/.htaccess

Switched to a new branch 'andig-vz.load-errors'

pi@raspberrypi /var/www/volkszaehler.org $ sudo git pull 
https://github.com/andig/volkszaehler.org.git vz.load-errors

remote: Counting objects: 74, done.
remote: Total 74 (delta 35), reused 35 (delta 35), pack-reused 39
Unpacking objects: 100% (74/74), done.
 From https://github.com/andig/volkszaehler.org
  * branch            vz.load-errors -> FETCH_HEAD
Updating 3bbeb9f..831ca61
Fast-forward
  htdocs/frontend/javascripts/entities.js         | 56 
+++++++++++++++++++++++++++-----------------------------
  htdocs/frontend/javascripts/entity.js           | 48 
++++++++++++++++++++++--------------------------
  htdocs/frontend/javascripts/functions.js        | 83 
+++++++++++++++++++++++++++++++++++++++++++----------------------------------------
  htdocs/frontend/javascripts/wui.js              | 54 
++++++++++++++++++++++++++----------------------------
  lib/Definition/PropertyDefinition.php           |  9 ++++++---
  lib/Interpreter/Interpreter.php                 | 16 +++++++++++++---
  lib/Interpreter/SQL/MySQLAggregateOptimizer.php | 39 
+++++++++++++++++++++++----------------
  lib/Interpreter/SQL/MySQLOptimizer.php          |  7 +++++++
  lib/Interpreter/SQL/SQLOptimizer.php            |  7 +++++++
  lib/Interpreter/SQL/SQLiteOptimizer.php         |  9 ++++++++-
  lib/Model/Entity.php                            |  6 ++++--
  lib/Model/Property.php                          |  4 ++--
  lib/Util/Debug.php                              | 25 
+++++++++++++++----------
  lib/View/JSON.php                               |  8 +++++++-
  lib/bootstrap.php                               |  8 +++++++-
  misc/tools/vzcompress2.php                      |  2 +-
  test/EntityTest.php                             | 18 ++++++++++++++++++
  17 files changed, 236 insertions(+), 163 deletions(-)

pi@raspberrypi /var/www/volkszaehler.org $

andig, 2015-12-30:
Was meinst Du mit

> Ok, bin nun wie folgt vorgegangen (bin ein DAU), konnte aber über das frontend keine Änderung sehen. 

Ist der Fehler immer noch vorhanden? Falls nein- läuft sonst alles soweit erkennbar?

donhasso, 2015-12-30:
Ahh, O.K.Hab nun versucht einen neuen Kanal "El. Energie (Zählerstände)" 
ohne eingabe der Auflösung zu erstellen und es kam eine Fehlermeldung, 
dass "resolution" fehlt, schaut also gut aus.

Ansonsten seh ich keine Auffälligkeiten.

andig, 2016-02-20:
Gut- schieb ich rein wenn ich wieder Zeit habe mögliche Fehler kurzfristig zu beheben. Die Änderung ist etwas zu gross für einen Schnellschuss.

PR #369 (2016-03-11): Implement jQuery promises for json error handling
Fix #363 - needs careful test!
closed 2016-03-11 by andig