volkszaehler/volkszaehler.org #367: Wunsch, Kanalattribut für sichtbar/unsichtbar

kr0815, 2015-12-10:
Ich habe eine rel. große Installation mit vielen Kanälen.
Diese ist organisiert in eine Hauptgruppe, viele Untergruppen.
Beim öffnen werden immer alle Kanäle geladen.
Wunsch wäre, wenn ich einstellen könnte, welche Kanäle geöffnet werden sollen.
Evtl. sogar so, falls einfacher, das erst mal gar kein Kanal aktiv ist, und ich diese aktiv schalten muss

andig, 2015-12-10:
Was spricht dagegen die Gruppen rein und raus zu nehmen? Wenn Du kein Cookie setzt startest Du ohnehin immer leer? Oder gibst Die gewünschten uuids über die Url an? Bitte Problem konkreter beschreiben.

kr0815, 2015-12-10:
Ich rufe die uuid der Hauptgruppe über eine URL / Weiterleitung auf, es gehen dann natürlich alle Untergruppen mit auf.
Ich will eigentlich nicht, das es komplett leer startet, sondern evtl. so, das die Häkchen vor den einzelnen Kanälen erst mal alle aus sind, oder evtl. beim Kanal ein zusätzlich Attribut "aktiv" gesetzt werden kann, das Frontend dann nur die die Werte für die aktiven Kanäle lädt.
Es sollen also schon alle Kanäle/Gruppen da sein, nur nicht aktiv.

Es gibt bei den Kanaleigenschaften auch ein "aktiv", hatte gedacht das macht genau das Gewünsche, leider aber nicht

andig, 2015-12-10:
Ruf doch die Untergruppen per Url auf? Die Anforderung ist aus meiner Sicht zu speziell.

kr0815, 2015-12-10:
schade, hatte ich mir schon gedacht

über URLs geht nicht, sonst müsste ich mir ja für alle Gruppen je eine Adresse merken und könnte nie 2 Kanäle aus unterschiedlichen Gruppen gleichzeitig anschauen?

Wozu ist denn die Auswahlbox "aktiv" bei den Kanaleigenschaften ? Vielleicht mache ich auch nur was falsch ?

jahir, 2015-12-14:
Hi! Ich kann die Anforderung schon nachvollziehen, da ich sowas selbst gerne hätte. Ich hatte mir überlegt, daß man über ein dropdown-Menü o.ä. verschiedene "Ansichten" auswählen kann. Bei mir wäre das z.B.
- Stromverbrauch im Haushalt
- Verbrauch/Betriebsdaten der Wärmepumpe
- PV Erzeugung/Einspeisung

Wobei man sich sowas natürlich auch bauen könnte, indem man für jede Ansicht eine Gruppe anlegen könnte. Idealerweise sollte dann beim initialen Laden des Frontends keine Gruppe geladen werden. Leider ist das Handling m.E. etwas inkonsistent: 
- Es gibt ein "sichtbar"-Flag im Frontend (der Haken links, unter dem Auge), das beim Starten wohl vom "active"-Flag aus der Middleware initialisiert wird. 
- Bei Gruppen ist das aber wirkungslos, da es nur beim Klicken auch auf die Mitglieder übertragen wird, aber nicht beim Laden. (Letzeres kann man wohl relativ einfach ändern.) 
- Leider wird das "sichtbar"-Flag nicht in den Cookies gespeichert (oder einfach von "active" aus der Middleware überschrieben), d.h. ein abschalten im Frontend wirkt nur bis zum nächsten reload. Das könnte (und sollte!) man wohl auch einfach ändern. Ist nur die Frage, wie man Konflikte zwischen FE und MW handhaben möchte.
- Premium wäre natürlich noch eine neue Gruppeneigenschaft, die aus den Häkchen Radio-Buttons machen würde. D.h., es wäre nur ein Mitglied der Gruppe auswählbar (meistens wäre das dann wohl  auch wieder eine von mehreren Gruppen, muß aber nicht sein). Ist aber ein bisschen mehr Arbeit und Umbau im Frontend und fraglich, ob sich das lohnt, weil...
- für richtig viele Channels/Gruppen/"Ansichten" ist das auch unhandlich, da wäre eine getrennte Auswahlbox oder so doch deutlich bequemer. Gibt aber wohl eher wenige, die so eine Anforderung haben

kr0815, 2015-12-14:
Verstehe gerade erst das Konzept hinter dem active-flag, hatte mich dadurch verwirren lassen das dieses je nach "sichtbar"-Stellung des Kanals mit verändert wird
Mir würde es reichen, wenn das Ganze auch für Gruppen funktionieren würde - genial wäre wenn bei inaktiver Gruppe diese beim Laden der Seite eingeklappt wäre.

Ich versuche das Ganze hier ohne Proxys zu machen, da ich auf vielen PCs zugreife

Viele Grüße

Klaus

andig, 2015-12-14:
> über URLs geht nicht, sonst müsste ich mir ja für alle Gruppen je eine Adresse merken und könnte nie 2 Kanäle aus unterschiedlichen Gruppen gleichzeitig anschauen?

Doch- Du kannst uuids auch als Array angeben (`uuid[]=abc&uuid[]=def`). Falls das nicht tut ist's schnell behoben.

> Wozu ist denn die Auswahlbox "aktiv" bei den Kanaleigenschaften ? Vielleicht mache ich auch nur was falsch ?

> Es gibt ein "sichtbar"-Flag im Frontend (der Haken links, unter dem Auge), das beim Starten wohl vom "active"-Flag aus der Middleware initialisiert wird.

War mir auch nicht klar.

> Bei Gruppen ist das aber wirkungslos, da es nur beim Klicken auch auf die Mitglieder übertragen wird, aber nicht beim Laden. (Letzeres kann man wohl relativ einfach ändern.) 

Check, das wäre schnell gemacht. Active der Gruppe würde dann Active des individuellen Kanals übersteuern, zumals das Flag auch durch show/hide nicht gespeichert wird. 
Wäre das das Wunschverhalten? Damit liessen sich identische Kanäle auch durch mehrere Gruppen steuern und damit ein/ausschalten.

> Leider wird das "sichtbar"-Flag nicht in den Cookies gespeichert (oder einfach von "active" aus der Middleware überschrieben), 

Check. Bei klick auf "show" wird nix an die MW geschickt und kein Cookie gesetzt. Es gilt als immer der per Kanalkonfiguration eingestellte Wert von `active`.

> d.h. ein abschalten im Frontend wirkt nur bis zum nächsten reload. Das könnte (und sollte!) man wohl auch einfach ändern. 

Warum? Einfach die Kanalkonfiguration ändern- das wäre out of the box.

> Ist nur die Frage, wie man Konflikte zwischen FE und MW handhaben möchte. 

Gar nicht. Wenn ich das Standardverhalten show/hide ändern möchte muss ich in den Kanal gehen und active setzen/löschen. Das wäre dann eine ganz schmale Lösung die zu 99% tickt wie heute.

> Verstehe gerade erst das Konzept hinter dem active-flag, hatte mich dadurch verwirren lassen das dieses je nach "sichtbar"-Stellung des Kanals mit verändert wird

Nein, genau das ist nicht der Fall, zumindest wird nichts an die MW geschickt.

> Mir würde es reichen, wenn das Ganze auch für Gruppen funktionieren würde - genial wäre wenn bei inaktiver Gruppe diese beim Laden der Seite eingeklappt wäre.

Sollte sich machen lassen.

> Ich hatte mir überlegt, daß man über ein dropdown-Menü o.ä. verschiedene "Ansichten" auswählen kann. 

Schöne Idee, hätte ich auch gerne- z.B. um auch Verbrauchswerte (z.B. Jahr mit Tagesverbräuchen) darzustellen. Ist per URL-Parametern im Prototyping für das Frontend hatte ich bisher keine gute Idee. PVlng zeigt wie man sowas ziemlich komplex machen kann.

andig, 2015-12-15:
Ich fasse mal zusammen: 

  - bei Start sollen inaktive Gruppen diese Eigenschaft vererben
  - alle Kinder sollen inaktiv sein
  - Gruppe soll kollabiert dargestellt werden
  - gleiche Kinder in anderer Gruppe oder alleine sollen sich separat konfigurieren lassen

Wenns das ist (und ohne an der Speicherung des active Status rumzufrickeln) habe ich einen ziemlich einfachen PR fertig, folgt sobald ich wider online kann.

kr0815, 2015-12-15:
Ich verstehe zwar nicht was Du mit dem letzten Punkt meinst - gleiche Kinder in anderer Gruppe- , aber sonst sieht es gut aus

  ----- Original Message ----- 
  From: andig 
  To: volkszaehler/volkszaehler.org 
  Cc: kr0815 
  Sent: Tuesday, December 15, 2015 12:18 PM
  Subject: Re: [volkszaehler.org] Wunsch, Kanalattribut für sichtbar/unsichtbar (#367)

  Ich fasse mal zusammen: 

    a.. bei Start sollen inaktive Gruppen diese Eigenschaft vererben 
    b.. alle Kinder sollen inaktiv sein 
    c.. Gruppe soll kollabiert dargestellt werden 
    d.. gleiche Kinder in anderer Gruppe oder alleine sollen sich separat konfigurieren lassen 
  Wenns das ist (und ohne an der Speicherung des active Status rumzufrickeln) habe ich einen ziemlich einfachen PR fertig, folgt sobald ich wider online kann.

andig, 2015-12-15:
Gemeint ist dass ein Kanal- wenn er in verschiedenen Gruppen steckt, oder auchmal ohne eine Gruppe einzeln- sich unabhängig verhalten soll. Es entscheidet jeweils die Gruppe oder- wenn in keiner Gruppe- die `active` Konfiguration des Kanals.

andig, 2015-12-15:
Bitte mal ausprobieren. 

Einziger Haken: der `active` Status eines Kanals ist undefiniert, je nachdem wo und in welcher Gruppe er zuletzt hängt. Wird der Kanal bearbeitet und gespeichert so wird dieser Status zum Server zurück geschrieben (allerdings _nur_ bei Speicherung). Das zu ändern wäre eine größere Aktion...

kr0815, 2015-12-15:
Hallo Andreas,

ich schreib Dir mal privat, will den Link nicht veröffentlichen

Habe hier die neue Version drauf gemacht, der Button "speichern" ist 
zwar da, aber speichern lässt sich nichts ?

esp.kr123.de

Darfst gerne auf dieser Installation rumklicken

habe einfach bei meiner Installation die Dateien durch die aus Github 
ersetzt, dann die volkszaehler.conf.php aus der vorherigen Version 
reinkopiert, müsste doch die richtige Vorgehensweise sein ?

Viele Grüße

Klaus

kr0815, 2015-12-21:
Habe es jetzt mal versuchen können - es funktioniert für die Hauptgruppe, aber nicht für Untergruppen?

andig, 2015-12-22:
> Kann es sein, das dann das aktiv-Attribut der Hauptgruppe übernommen wird?

Jaein. Die Logik war etwas krude. Versuch bitte mal folgendes in der entity.js->inheritVisibility Funktion:

		vz.entities.inheritVisibility = function() {
		vz.entities.each(function(entity, parent) {
			console.log(entity.title + " " + entity.active);
		}, true);
		console.log("--");

			vz.entities.each(function(entity, parent) {
				console.log(entity.title + " " + entity.active);

				// inherit active state if parent
				if (entity.type !== "group" && entity.parent !== undefined) {
		console.log("   -> from parent (" + parent.title + ") " + entity.parent.active);
					// entity.active = entity.parent.active;
					if (entity.active !== entity.parent.active) {
						entity.activate(entity.parent.active);
					}
				}

				// collapse groups if inactive
				if (entity.type == "group" && entity.active === false) {
		console.log("   -> collapsing");
                                        entity.activate(false, entity.parent, true);
					$('#entity-' + entity.uuid + '.aggregator').removeClass('expanded').collapse();
				}
			}, true);
		};

Die Logik damit wäre jetzt also:

  - wenn keine Gruppe dann `active` von übergeordneter Gruppe erben wenns eine gibt
  - wenn Gruppe und nicht `active` dann einklappen und alles untergeordnete deaktivieren

Das eingebaute Logging zeigt Dir in der JS Konsole an was und warum da gerade passiert.
Wenn das auch nicht passt musst Du mal beschreiben wie es "wirklich" sein soll ;)

kr0815, 2015-12-23:
Danke Andreas,

jetzt sieht es besser aus, die einzelnen Untergruppen lassen sich als aktiv / nicht aktiv speichern und werden beim neu laden so übernommen.
Habe es aber in entities.js geändert, nicht entity.js

2 kleine Schönheitsfehler gibt es noch:

- zum Anzeigen einer inaktiven Gruppe nachdem die Seite neu geladen wurde bedarf es 2 Mausklicks - ausklappen und aktivieren
- nach ändern aktiv / inaktiv einer Gruppe und speichern klappen alle anderen Gruppen auf, ist aber nicht tragisch, reload der Seite reicht dann

Viele Grüße

Klaus

----- Original Message ----- 
  From: andig 
  To: volkszaehler/volkszaehler.org 
  Cc: kr0815 
  Sent: Tuesday, December 22, 2015 1:14 PM
  Subject: Re: [volkszaehler.org] Wunsch, Kanalattribut für sichtbar/unsichtbar (#367)

    Kann es sein, das dann das aktiv-Attribut der Hauptgruppe übernommen wird?

  Jaein. Die Logik war etwas krude. Versuch bitte mal folgendes in der entity.js->inheritVisibility Funktion:

    vz.entities.inheritVisibility = function() {
    vz.entities.each(function(entity, parent) {
        console.log(entity.title + " " + entity.active);
    }, true);
    console.log("--");

        vz.entities.each(function(entity, parent) {
            console.log(entity.title + " " + entity.active);

            // inherit active state if parent
            if (entity.type !== "group" && entity.parent !== undefined) {
    console.log("   -> from parent (" + parent.title + ") " + entity.parent.active);
                // entity.active = entity.parent.active;
                if (entity.active !== entity.parent.active) {
                    entity.activate(entity.parent.active);
                }
            }

            // collapse groups if inactive
            if (entity.type == "group" && entity.active === false) {
    console.log("   -> collapsing");
                                    entity.activate(false, entity.parent, true);
                $('#entity-' + entity.uuid + '.aggregator').removeClass('expanded').collapse();
            }
        }, true);
    };
Die Logik damit wäre jetzt also:

    a.. wenn keine Gruppe dann active von übergeordneter Gruppe erben wenns eine gibt 
    b.. wenn Gruppe und nicht active dann einklappen und alles untergeordnete deaktivieren 
  Das eingebaute Logging zeigt Dir in der JS Konsole an was und warum da gerade passiert.
  Wenn das auch nicht passt musst Du mal beschreiben wie es "wirklich" sein soll ;)

andig, 2015-12-23:
> - zum Anzeigen einer inaktiven Gruppe nachdem die Seite neu geladen wurde bedarf es 2 Mausklicks 

Denke das ist ok- sind ja auch 2 verschiedene Aktionen. Anklicken der Box aktiviert auf jeden Fall die Unterlemente auch wenn eingeklappt.

> - ausklappen und aktivieren - nach ändern aktiv / inaktiv einer Gruppe und speichern klappen alle anderen Gruppen auf, ist aber nicht tragisch, reload der Seite reicht dann

Das ist das originale Verhalten- aktivieren ist rekursiv wenn man die Listenansicht verwendet.

Würde ich Beides so lassen.

andig, 2015-12-23:
PR aktualisiert inkl. Fehlerbehebung beim Neustart- bitte testen.

andig, 2015-12-28:
Merged after successful test by Klaus.

closed 2015-12-28 by andig

justinotherguy, 2016-01-07:
...habe mich gerade darüber gefreut, dass das eingebaut ist - sehr schön :)