volkszaehler/volkszaehler.org #370: volkszaehler.org should use "real" ssl certificate

andig, 2015-12-27:
Any chance to move to http://letsencrypt.org?

justinotherguy, 2015-12-29:
A good idea - major advantage from my perspective: more browsers than cacert.
Do you know, whether this can be used in fully manual "old fashioned CA mode"?
Whereas the fully default automatic mode is excellent for unexperienced admins, I don't really like the idea of having code run for this purpose.
I was unable to answer the question by checking letsencrypt's website.

andig, 2015-12-29:
It is. I seem to recall a ct (or web?) article on the subject. It doesn't make sense though as the certs are shortlived (I believe 2 months currently). I'll check if I can find the source.

schnello, 2016-02-20:
Hallo.

Also so schlecht ist "http://letsencrypt.org" im Grunde nicht. Die Zertifikate selbst halten zwar nur 3 Monate. Es gibt jedoch eine "auto update" Funktion. Wie gut oder schlecht die Arbeitet kann ich dann in 60 Tagen sagen.

Alternativ gibts dann noch https://www.startssl.com/
Sollte 1 Jahr sein... aber dafür braucht man Zugriff auf die postmaster EMail Adresse der Domain.

Grüße

andig, 2017-02-09:
@justinotherguy implemented letsencrypt.

closed 2017-02-09 by andig