volkszaehler/volkszaehler.org #375: Website: install pushserver on demo

andig, 2015-12-29:
Either according to ancient ML posts (need to dig) or I'll volunteer.

@sollner could you post tutorial here?

Sollner, 2015-12-30:
Seit dem Raspberry Pi image-Stand 25.11.2015 von Udo 
https://demo.volkszaehler.org/downloads/volkszaehler_latest.rar 
ist es möglich, Daten im Frontend anzuzeigen, die nicht aus der Datenbank kommen.

Das hört sich unspektakulär an, ist es aber nicht.
Es ist ein langgehegter Menschheitswunsch ;-).

Es werden Daten dargestellt, die sozusagen viel höher aufgelöst sind, als in interval oder aggtime angegeben. Zugegeben, wer schon immer eine aggtime von 2 Sekunden hat, der wird keinen Unterschied darstellen. Wer legt aber so exorbitant viele Daten in die DB ab.

Mit den Live-Daten kann man nun per Frontend auf die Pirsch gehen, und auch kleinste Verbräuche aufspüren.

Aktuell basiert das image für einen Raspberry Pi von Udo auf Jessie und nicht mehr Wheezy
dadurch sind einige Sachen anders, z.B. die Art und Weise, wie vzlogger gestartet wird

1. upgrade vzlogger to volkszaehler/vzlogger#174
wir sollen den vzlogger aktualisieren:
http://wiki.volkszaehler.org/software/controller/vzlogger/installation_cpp-version
am besten auch gleich das ganze System:
Paketliste aktualisieren
```sudo apt-get update```
Pakete herunterladen und installieren
```sudo apt-get upgrade```

Für einen aktuellen VZ:
Composer aktualisieren
```sudo /usr/local/bin/composer self-update```
Ins Verzeichnis wechseln
```cd /var/www/volkszaehler.org```
Volkszähler aktualisieren
```sudo git pull```
Abhängigkeiten auflösen
```sudo composer update```

2. enable push server in volkszaehler.conf.php
nun muss der pushserver aktiviert werden
das passiert in der o.g. Datei volkszaehler.conf.php
wir schauen da mal rein:
```nano /var/www/volkszaehler.org/etc/volkszaehler.conf.php```
ok, einfach: false durch true ersetzen, Rest bleibt
```/**
 * Push server settings
 */
$config['push']['enabled'] = true;              // set to true to enable push updates
$config['push']['server'] = 5582;               // server will always listen on 0.0.0.0
$config['push']['broadcast'] = 8082;    // server will always listen on 0.0.0.0```

3. make sure external port is accessible to internet (if required)
für Zugriff von aussen muss der entsprechende Port weitergeleitet werden
dazu im Router den Port 8082 von extern auf den 8082 am Raspberry Pi weiterleiten

4. start push server
genau wie der vzlogger bei Jessie als Dienst gestartet wird, wird im neuen Image auch der pushserver als Dienst eingerichtet
wer nicht das neue Image hat, macht das selber
dazu einfach die Datei vzlogger.service zu pushserver.service umkopieren
```cp /etc/systemd/system/vzlogger.service /etc/systemd/system/pushserver.service```
und dann den Inhalt austauschen gegen:
```[Unit]
Description=pushserver
After=syslog.target network.target
Requires=mysql.service

[Service]
ExecStart=/usr/bin/php /var/www/volkszaehler.org/misc/tools/push-server.php
ExecReload=/bin/kill -HUP $MAINPID
StandardOutput=null
Restart=always

[Install]
WantedBy=multi-user.target```
danach ein 
[... 49 lines cut ...]

entweder lässt man den Dienst immer laufen, oder startet ihn wie oben beschrieben

noch ein paar Erläuterungen:

Der pushdienst muss laufen,
vzlogger schickt an Dienst (5582), 
Dienst schickt an alle frontends, 
frontend meldet sich beim dienst an (8082).

vzlogger.conf:
Die ganze Sache ist nicht mit dem eingebaute HTTPD Port 8082 vom vzlogger zu verwechseln, der Port ist nur zufällig der gleiche, darf aber bei Verwendung eben dann nicht Port 8082 sein

in der .conf ist per ```nano /etc/vzlogger.conf``` einzutragen:
```"push": [
      {
      "url": "http://127.0.0.1:5582"
      }
    ],

    // Build-in HTTP server
    "local" : {
        "enabled" : false,       /* should we start the local HTTPd for serving live readings? */
        "port" : 8082,          /* the TCP port for the local HTTPd */
'''

Sollner, 2015-12-30:
@andig ... done ;-)

justinotherguy, 2015-12-30:
Das liest sich gut - diese 3 Fragen stellen sich mir beim Lesen:
- muss das wirklich der gleiche Port für den Push-Server sein? Das schreit doch nach Ärger, oder?
  Vorschlag: lasst uns da einen anderen nehmen (es gibt ja noch ein paar freie...) wie wär's mit 8083?
- welche Rollen spielen die Ports 5582 und 8082 im Push-Kontext? wer kommuniziert in welchem Fall mit denen?
- worauf baut der Code der beiden zusätzlichen Web-Server? Wie groß ist die Chance, dass wir uns damit sicherheitstechnisch ein Ei legen? Im LAN ist das ja wurscht (solange das nicht aus dem Internet erreichbar ist) - aber auf demo.vz stelle ich mir die Frage schon... unter welchen Usern laufen die beiden Server?
Insbesondere bei der 3. Frage lohnt es sich wohl, einen Moment inne zu halten - wenn wir da ne gute Antwort haben, aktiviere ich das gerne.
Gruß, J.

andig, 2015-12-30:
> Das liest sich gut - diese 3 Fragen stellen sich mir beim Lesen:

Ich dübele das Zeugs ja nur zusammen- deshalb bin ich sehr froh dass Du die richtigen Fragen stellst!

Was Patric vergessen hat zu erwähnen ist die Architektur, siehe auch https://github.com/volkszaehler/volkszaehler.org/blob/master/etc/volkszaehler.conf.template.php#L104

Der Pushdienst ist Teil der Middleware. Der Dienst muss laufen, vzlogger schickt an Dienst (5582), Dienst schickt an alle Frontends wenn sich Frontend vorab beim Dienst angemeldet hat (8082).

>   - muss das wirklich der gleiche Port für den Push-Server sein? Das schreit doch nach Ärger, oder? Vorschlag: lasst uns da einen anderen nehmen (es gibt ja noch ein paar freie...) wie wär's mit 8083?

Siehe oben. Es braucht 2 Ports:

	/**
	 * Push server settings
	 */
	$config['push']['enabled'] = false;             // set to true to enable push u$
	$config['push']['server'] = 5582;               // server will always listen on$
	$config['push']['broadcast'] = 8082;    // server will always listen on 0.0.0.0

Welche ist egal, die 8082 ist in der options.js des Frontends voreingestellt.

>   - welche Rollen spielen die Ports 5582 und 8082 im Push-Kontext? wer kommuniziert in welchem Fall mit denen?

Siehe oben.

> - worauf baut der Code der beiden zusätzlichen Web-Server? Wie groß ist die Chance, dass wir uns damit sicherheitstechnisch ein Ei legen? Im LAN ist das ja wurscht (solange das nicht aus dem [...]

`push-server.php` ist ein Service in PHP der keinen Webserver benötigt. 
Der interne wie externe Port werden an `0.0.0.0`gebunden. Intern um auch remote vzloggern das 'pushen' zu ermöglichen. Wen vzlogger lokal läuft kann dieser Port jedoch intern bleiben.
Kommunikation extern erfolgt über WAMP. Serverseitig mittels https://packagist.org/packages/cboden/ratchet mit 500tsd Installationen. Der Post muss sichtbar sein damit das Frontend connecten kann.

Ich bin mir noch nicht abschließend sicher ob das aus Demo- da https- überhaupt funktioniert da damit auch WSS Sockets erforderlich werden. Ohne Test werden wir das allerdings auch nicht rausfinden :/

andig, 2016-01-08:
Schau mal Justin- hier ist sogar noch das Diagramm von unserer hacking session: https://github.com/volkszaehler/volkszaehler.org/issues/219

Fehlt Dir noch was (ausser Zeit)?

justinotherguy, 2016-01-09:
Ok, done :-)

Hab aber ein paar Anpassungen vorgenommen:
- der Server sollte nicht als root gestartet werden
  - bei mir (inittab, weil noch nicht systemd-basiert):
    # spawn push server
    push:235:respawn:/bin/su - www-data -c "/usr/bin/php /var/www/htdocs/kjm/demo.volkszaehler.org/misc/tools/push-server.php"
  - für systemd-basierte Systeme sollte im Abschnitt "Service" ergänzt werden (noch zu prüfen):
    User=www-data
- obiger Kommentar zum Port des lokalen httpd von vzlogger ist verwirrend - der läuft standardmäßig auf Port 8080, nicht auf 8082:
https://github.com/volkszaehler/vzlogger/blob/master/etc/vzlogger.conf#L23
Da gibt's also gar keine Kollision :)
Ansonsten: bitte testen, auf demo.vz läuft er nun :)

closed 2016-01-09 by justinotherguy

andig, 2016-01-09:
> obiger Kommentar zum Port des lokalen httpd von vzlogger ist verwirrend - der läuft standardmäßig auf Port 8080, nicht auf 8082:

Ich habs jetzt 3x gelesen- welcher Kommentar?

Sollner, 2016-01-09:
hatte den gerade rausgenommen, deswegen kannst du ihn nicht mehr lesen

Patric