volkszaehler/volkszaehler.org #378: CSV-Ausgabe aus dem Frontend heraus: Falscher Dateiname und Ausgabe in Unix-Time

UdoSchake, 2016-01-03:
Wenn ich am 03. Jan. 2016, 15:45Uhr aus dem Frontend heraus über die Export-Funktion eine CSV-Ausgebe heraus mache erhalte ich:
1. einen falschen Dateinamen ohne Endung: d0.csv- creation-date=-Thu, 01 Jan 1970 01-00-00 +0100
2. sind die Zeiten im File als Unix-Time aufgeführt.:

<code>
# source:;volkszaehler.org
# version:;0.3

# uuid:;2e140e40-6ac1-11e5-93f4-25f19fcc1e97
# title:;28-000004508eec
# from:;1451832047936
# to:;1451832117162
# min:;1451832117162; => ;21.75
# max:;1451832065366; => ;22.312
# average:;22.092
# rows:;5
1451832065366;22.312;1
1451832082302;22.312;1
1451832099733;22;1
1451832117162;21.75;1

# uuid:;46a75c40-6abe-11e5-87fb-119733289a5c
# title:;28-000004508eec
# from:;1450725840618
# to:;1450725878900
# min:;1450725878900; => ;22.2
# max:;1450725878900; => ;22.2
# average:;22.2
# rows:;2
1450725878900;22.2;1

# uuid:;630cf350-6abe-11e5-ad9c-1fd5012e73e3
# title:;28-00000450a4cc
# from:;1451832063746
# to:;1451832115543
# min:;1451832115543; => ;21.25
# max:;1451832080683; => ;21.937
# average:;21.601
# rows:;4
1451832080683;21.937;1
1451832098112;21.625;1
1451832115543;21.25;1

# uuid:;84f2ac40-87c7-11e5-8530-2b84c86eab69
# title:;28-00000450b617
# from:;1450725840601
# to:;1450725878869
# min:;1450725878869; => ;22.2
# max:;1450725878869; => ;22.2
# average:;22.2
# rows:;2
1450725878869;22.2;1

# uuid:;77ea6690-6abe-11e5-885f-2b5eba09e0c2
# title:;28-00000450cbbd
# from:;1451832064555
# to:;1451832116352
# min:;1451832116352; => ;21.625
# max:;1451832081492; => ;22.25
# average:;21.956
# rows:;4
1451832081492;22.25;1
1451832098922;22;1
1451832116352;21.625;1

# uuid:;e2032bf0-6abf-11e5-ad05-958ad22209ff
# title:;S0/0
# from:;1451831942814
# to:;1451832104520
# min:;1451832092524; => ;30.652
# max:;1451832104520; => ;300.1
# average:;66.788
# consumption:;3
# rows:;4
1451831975075;111.59;1
1451832092524;30.652;1
1451832104520;300.1;1
snip....
</code>

Wünschenswert wäre ein Dateiname mit Dateiendung und die Zeitangaben in CET.

Gruß
Udo

Ps: was ist denn das für eine sch...-Formatierung hier?

andig, 2016-01-03:
Hi @UdoSchake,

> 1. einen falschen Dateinamen ohne Endung: d0.csv- creation-date=-Thu, 01 Jan 1970 01-00-00 +0100

Bei mir heisst der bezug.csv. Welchen Browser hast Du?

> 2. sind die Zeiten im File als Unix-Time aufgeführt.:

Ich hatte Dir schon per PM geschrieben wie Du das Format ändern kannst und vorgeschlagen das als Standard bei CSV zu machen. Ist es das was Du willst?

Ich bau ja alles, aber Du musst mir antworten ;)

> Ps: was ist denn das für eine sch...-Formatierung hier?

Keine Ahnung ob GH Markdown <code> kann- wenn Du's 4 spaces einrückst siehts gut aus (alternativ mit ``` einschließen).

UdoSchake, 2016-01-03:
Hallo Andy,
Mit Chrome ausprobiert. Gerade mit Firefox ausprobiert, da bekomme ich auch die Endung .csv.
Sorry, ich hatte nicht ganz verstanden was du mit der PM-Mail gemeint hast.
Gruß
Udo

andig, 2016-01-09:
Fixed

closed 2016-01-09 by andig