volkszaehler/volkszaehler.org #379: composer packages

milefork, 2016-01-04:
Nabend,

erstmal wünsche ich euch ein frohes Neues und ein großes Lob an @andig das du so fleißig develst.

Ich habe gerade mal ein git pull & composer install gemacht und dabei hat composer das hier ausgespuckt:

Package guzzle/common is abandoned, you should avoid using it. Use guzzle/guzzle instead.
Package guzzle/stream is abandoned, you should avoid using it. Use guzzle/guzzle instead.
Package guzzle/parser is abandoned, you should avoid using it. Use guzzle/guzzle instead.
Package guzzle/http is abandoned, you should avoid using it. Use guzzle/guzzle instead.

Ich weiß das ganze ist alles andere als tragisch, nur sollte man sich nicht überlegen dann zu dem Paket zu wechseln? Ich denke mal das andere wird nicht mehr weiterentwickelt (so aus der bugfix und security Sicht)?

Gruß

andig, 2016-01-04:
Hi Christian,

> 
> Nabend,
> 
> erstmal wünsche ich euch ein frohes Neues und ein großes Lob an @andig das du so fleißig develst.

Danke :)
> Ich habe gerade mal ein git pull & composer install gemacht und dabei hat composer das hier ausgespuckt:
>
> Package guzzle/common is abandoned, you should avoid using it. Use guzzle/guzzle instead. [...]
Ist bekannt und erstmal kein Problem. Irgendwann werd ich wechseln, momentan geht das aber nicht wg cboden/ratchet wenn ich mich mich richtig erinnere. Hab ich aber auf dem Radar!
> Gruß
Viele Grüße, Andreas

closed 2016-01-08 by andig

justinotherguy, 2016-12-11:
Blöde Frage:
hast Du ne Idee, wie wir mit den guzzle-Paketen umgehen?
Steht da zu befürchten, dass die Warning mal zu nem Error wird?
Falls wir auf absehbare Zeit damit leben können: sollten wir den Issue mal offen lassen als Erinnerung?

Ansonsten kann ich mich mileforks Lob nur anschliessen :)

Gruß, J.

andig, 2016-12-11:
Das ist harmlos. Lösung sobald die Upstream Bibliothek auf neues Guzzle umstellt.

Viele Grüße, Andreas