volkszaehler/volkszaehler.org #381: Live updates: table updated but not chart

andig, 2016-01-13:
Reproducible @sollner11

andig, 2016-06-25:
Additional issue: current period consumption not updated on live updates.

andig, 2017-04-13:
cc @frankrichter for 2nd issue. First issue is wfm.

frankrichter, 2017-04-13:
Genau, nicht aktualisierten Chart hatte ich noch nie, aber laufende Verbrauchsanzeige bei Push wär ne feine Sache.
Dann stellt sich noch die Frage: was machen mit min/max? Aktuell berechnen stell ich mir zu aufwändig vor, aber vielleicht wäre Darstellung in grau (für möglicherweise veraltete Werte) sinnvoll.

frankrichter, 2017-04-20:
consumption update gibt es schon, ist nur nicht ganz richtig: https://github.com/volkszaehler/volkszaehler.org/blob/master/htdocs/js/entity.js#L224
wenn ich `this.consumption` zu `this.data.consumption` ändere, wird der Verbrauch brav aktualisiert.

Allerdings sollte die Berechnung von consumption eigentlich nicht von initialconsumption abhängig sein, und es fehlt eine Korrektur für die nach links aus dem Intervall rausgeschobenen Tupel.

andig, 2017-04-20:
> es fehlt eine Korrektur für die nach links aus dem Intervall rausgeschobenen Tupel.

fehlt noch

closed 2017-04-25 by andig