volkszaehler/volkszaehler.org #382: Live updates: investigate if port 80 can be used instead of 8082

andig, 2016-01-13:


andig, 2016-02-11:
This is possible. We need to add the following to `http.conf`:

	LoadModule proxy_module modules/mod_proxy.so
	LoadModule proxy_wstunnel_module modules/mod_proxy_wstunnel.so

	<Location "/volkszaehler.org/ws">
	    ProxyPass ws://localhost:8082
	</Location>

	<Location "/volkszaehler.org/wss">
	    ProxyPass wss://localhost:8082
	</Location>

As this approach doesn't need another port it would be the preferred solution. The `middleware.live` parameter would still be available if separate port is required.

Tested on Apache 2.4.

ping @justinotherguy could you add this config to demo.volkszaehler.org? Both locations would be required to allow http plus https.

justinotherguy, 2016-02-11:
ok, done - pls check, if this works - thx!

closed 2016-02-13 by andig