volkszaehler/volkszaehler.org #383: Messwertabfrage mit tuples=0

J-A-U, 2016-01-18:
Laut Doku http://wiki.volkszaehler.org/development/api/reference#daten-kontext soll der Parameter &tuples=[Integer] die Anzahl an zurückgegebene Datenpunkten eingrenzen. Bei =0 sollen keine Punkte sondern nur die Ergnbnisse für die Tabelle im Frontend unten zurück kommen (min, max, consumption).
Dem ist aber nicht der Fall. Die Middleware antwortet mit allen Datenpunkten im angeforderten Zeitrahmen.

andig, 2016-01-18:
Beispielaufruf? Gerne gegen demo damit es nachvollziehbar wird.

J-A-U, 2016-01-18:
demo.volkszaehler.org/middleware.php/data/57acbef0-88a9-11e4-934f-6b0f9ecd95a8.json?from=7 days ago&tuples=0

andig, 2016-01-18:
> http://demo.volkszaehler.org/middleware.php/data/57acbef0-88a9-11e4-934f-6b0f9ecd95a8.json?from=7 days ago&tuples=0

Erstmal sollten bei HTTP die ` ` eigentlich `+` sein, das geht aber meist trotzdem. Falsch ist `tuples=0`, mit `tuples=1` geht es. Schön wäre natürlich eine Fehlermeldung...

J-A-U, 2016-01-18:
Im Wiki steht:
"für &tuples=0 werden nur Max, Min & Durchschnitts-Werte berechnet"
Ob das irgendwann mal so funktioniert hat kann ich aber nicht sagen.

OT: Die Leerzeichen sind ein C&P-Fehler. Normal macht Opera da ein %20 draus. In dem Fall seltsamerweise nicht.

andig, 2016-01-19:
Du darfst das Wiki gerne korrigieren. 

Viele Grüße,
Andreas

andig, 2016-02-11:
> Du darfst das Wiki gerne korrigieren.

Hat ja nix mehr mit dem Issue zu tun, ich machs also zu.

closed 2016-02-11 by andig