volkszaehler/volkszaehler.org #384: Virtuelle Kanäle

J-A-U, 2016-01-18:
Einer von Andis Testzweigen hatte mal zusätzliche, virtuelle Kanäle die über eine Formel in den Kanalparameter aus anderen Kanälen errechnet wurden.

Bitte implementiert diese Funktion im Hauptzweig.

andig, 2016-01-18:
Grosser Rewrite notwenig, momentan leider wenig Zeit. Kein "Issue"...

J-A-U, 2016-01-18:
Verständlich. Ich dachte halt ich nehm mal den "offiziellen" Weg.

Was wäre denn die richtige "Art" um solcherlei Wünsche zu äußern?

andig, 2016-01-18:
Der Meldeweg ist schon ok für Featurerequests. An der Zeit ändert das leider nix.

mklhmnn, 2016-04-10:
Ja, gerade mit PV-Anlagen wäre es toll, z.B. den echten Hausverbrauch (PV-Erzeugt - Verkauft + Gekauft) anzeigen lassen zu können.

andig, 2016-04-10:
Ich würde zusätzlich gerne noch eine Saldierung einbauen, also z.B um Tages- und Jahresproduktion aufzusummieren und als Kurve darzustellen. Leider ist das ein ziemlich aufwändiges Unterfangen...

Saftwerk, 2016-05-28:
Ein "Virtueller Kanal" sollte gar keinen Platz in der Datenbank brauchen. Man kann die Werte vom Frontend aus den anderen abonnierten Werten berechnen lassen. Im Prinzip ist das dann wie eine Gruppe mit zusätzlicher Formel.

andig, 2017-01-26:
Update: wer ganz mutig ist ohne keine Angst vor Schmerzen hat schaut mal hier rein: https://github.com/andig/volkszaehler.org da gibts auch virtuelle Kanäle in einer allerersten Testversion.

TMHEINE, 2018-06-29:
Ist ja jetzt schon ne Weile her...
Gibt es inzwischen schon neueres oder Lösungsansätze?
Als PV-Besitzer wäre das wirklich super, um den Aktualverbrauch darzustellen (Erzeugung-Verkauf+Bezug=Aktualverbrauch)

frankrichter, 2018-06-29:
Im verlinkten Fork von andig sind virtuelle Kanäle implementiert. Ich
benutze das seit über einem Jahr im Normalbetrieb. Einfach mal ausprobieren!

closed 2019-03-21 by andig