volkszaehler/volkszaehler.org #387: Fehler "Autobahn not connected" beim ein- und ausschalten von Kanälen vom Typ "Betriebsstundensensor" im Frontend

justinotherguy, 2016-02-03:
Hi,
ich mach das:
- Safari 9.0.3 unter OSX starten
- Zugriff auf http://demo.volkszaehler.org (keine Kanäle abonniert, Cookies gelöscht)
- Kanal hinzufügen -> Kanal erstellen -> Betriebsstundensensor, Titel "test", Auflösung 1 -> erstellen
-> Kanal wird im Frontend aufgelistet
- Kanal deaktivieren -> Javascript Runtime Error: "Autobahn not connected"
Aufgetreten auch mit 
- Firefox 43.0.4 und Chromium unter FreeBSD
- Chrome 48.0.2564.97 (64-bit) (Meldung dort: "Uncaught Autobahn not connected")

nicht mit FF 44 unter OSX

$LIEBLINGSSUCHMASCHINE findet:
http://www.photovoltaikforum.com/volkszaehler-org-f131/raspberry-pi-2-volkszaehler-vs-ehm-dmtz-t110636.html
http://www.photovoltaikforum.com/volkszaehler-org-f131/kanaele-in-fronted-ueber-wlan-anlegen-t110656.html
https://github.com/greenaddress/WalletCordova/issues/20
$MAINSTREAMSUCHMASCHINE findet noch mehr

Gruß, J.

andig, 2016-02-07:
Kann ich mit Win/FF reproduzieren. Den hier hast Du schon drin https://github.com/volkszaehler/volkszaehler.org/commit/2164482d8e71056c20a82ed2967457baad4d9598 ?

Falls ja würd ich mich gerne per ssh auf demo umschauen.

bgewehr, 2016-02-07:
Nutzen wir autobahn.js?

https://www.google.de/search?q=Uncaught+Autobahn+not+connected&ie=UTF-8&oe=UTF-8&hl=de-de&client=safari

Gruß
Bernd

> Am 03.02.2016 um 09:28 schrieb Justin Otherguy <notifications@github.com>:
> 
> Uncaught Autobahn not connected

andig, 2016-02-07:
Ja, wieso?

andig, 2016-02-10:
> Den hier hast Du schon drin 2164482 ? Falls ja würd ich mich gerne per ssh auf demo umschauen.

ping @justinotherguy

justinotherguy, 2016-02-11:
> Den hier hast Du schon drin 2164482 ? 

ja, da läuft 38c648d

> Falls ja würd ich mich gerne per ssh auf demo umschauen.

gerne

> Eigentlich sollte auf demo auch der pushserver laufen- das scheint nicht mehr der Fall zu sein? 

doch:
netstat -tlnep | grep php
tcp        0      0 0.0.0.0:5582            0.0.0.0:*               LISTEN      33         7467        3009/php
tcp        0      0 0.0.0.0:8082            0.0.0.0:*               LISTEN      33         7483        3009/php

> Für die > Fehlerdiagnose aber erstmal besser so.

soll ich den Push-Server ausschalten?

andig, 2016-02-15:
Solved by https://github.com/volkszaehler/volkszaehler.org/pull/405

closed 2016-02-15 by andig