volkszaehler/volkszaehler.org #389: Neuen Kanaltyp "Wert (ohne Einheit)" einführen?

justinotherguy, 2016-02-04:
Hi,
da das alles so prächtig funktioniert, verwende ich MW und FE inzwischen auch für ganz andere Dinge:
- verbleibende MB im gebuchten Traffic-Paket auf der SIM-Karte (Einheit: MB)
- Restguthaben auf dem SIM-Kartenkonto (Einheit: EUR)
- Anzahl Mitglieder auf einer ML
Da gibt es bestimmt weitere Anwendungsfälle. Für jeden Datentyp nun einen eigenen Kanaltyp zu definieren, erscheint mir nicht sinnvoll - einen zusätzlichen universellen Kanaltyp hingegen schon.
Der könnte dann entweder ganz ohne Einheit sein (die kann man dann ja im "title" eintragen) oder vielleicht sogar als Parameter die Eingabe einer Einheit erlauben.
Meinungen?
Gruß, K.

andig, 2016-02-04:
Kein Problem, aber als was? Sensor, Zählerstand oder Impulse?

justinotherguy, 2016-02-05:
Ich nutze bisher dafür "El. Energie (Leistungswerte)";
Ziel wäre, absolute Werte zu loggen; drum sind Impulse raus; 
Sensor dürfte auch nicht passen - Zählerstände, also, oder?

andig, 2016-02-07:
> Ziel wäre, absolute Werte zu loggen; drum sind Impulse raus; Sensor dürfte auch nicht passen - Zählerstände, also, oder?

Ich denk Du bräuchtest Beides- Meter und Sensor. Muss das Ding "Verbäuche" können oder nur Graphen?

KlausStetter, 2016-02-11:
Das hätte ich auch gerne!
Ich logge z.B. CO2-Konzentration in ppm, für die es schlicht nicht den richtigen Zählertyp gibt.
Ein universeller, am besten mit optional konfigurierbarer Einheit, wäre optimal.

andig, 2016-02-12:
Moin,

Einen neuen Typ anzulegen dauert mit Texteditor 30 Sekunden- einfach selber einbauen? Gerne auch als PR zur Verfügung stellen. CO2 erscheint mir auch sinnvoll- deutlich mehr als "Radioaktivität" die wir schon haben...

Viele Grüße, Andreas

KlausStetter, 2016-02-12:
Ok, hab das json-file gefunden, Danke.

dervitti, 2016-02-28:
#411: Add Co2 and Fan rpm entity description and one new icon für co2

andig, 2016-06-25:
VZ meeting: wird implementiert. Icon?

andig, 2016-09-26:
@all Ich habe jetzt auch die Möglichkeit hinzu gefügt, für einen "universellen" Sensor eine Unit zu definieren.

** Könnte jemand den PR testen? **

closed 2016-10-11 by andig