volkszaehler/volkszaehler.org #390: Additional view "value"

justinotherguy, 2016-02-04:
Servus,
the frontend is brillant to visualize data series over time, analyse data for a specific time frame, read average/max/min values etc.

For some use cases I think it's very interesting to retrieve only a single reading of the latest value:
- the current power
- the current temperature
This would be very handy for small displays (think: display for power situated somewhere in the living room that only shows "1,416 Watts" or "21.4 °C".
The easiest approach for this that I can think of would be a new "view" type that could be named "value" on top of the ones we have already: https://github.com/volkszaehler/volkszaehler.org/tree/master/lib/View.

To retrieve the value you would then access e.g. 
https://demo.volkszaehler.org/middleware.php/data/<uuid>.value
and display the result (maybe we could allow a parameter "?append_unit=yes" or such, so the result would not be "1,416", but "1,416 Watts".

Picture this:
- esp8266
- 1" OLED display (http://www.ebay.de/itm/OLED-0-96-White-128X64-I2C-IIC-Display-Module-U8GLIB-ESP8266-Arduino-UK-Tested-/261831125278)
packed together in a plastic enclosure (http://blog.squix.ch/2015/06/esp8266-3d-printed-case-for-weather.html or http://uk.rs-online.com/web/p/general-purpose-enclosures/4732026/ or maybe http://www.ebay.de/itm/Boitier-Aluminium-DIY-Aluminum-enclosure-shell-70x24x80-Pi-STM32-ESP8266-PIC-RF-/272072076560 or even https://learn.adafruit.com/huzzah-weather-display/enclosure-assembly)

Could be used for creating "home screen icons" for smart phones to get quick access to the current values, too.

Nice, eh?

Regards, J.

andig, 2016-02-04:
You can do this this 5 lines of js already?

andig, 2016-02-04:
Or rather: yes I like it! Lets make it a ".plain" context and remove xml in turn.

justinotherguy, 2016-02-05:
.plain: nice!
remove .xml: agreed - maybe we should ask or at least notify quickly on the ML
I didn't understand your comment with js, though

closed 2016-02-14 by andig