volkszaehler/volkszaehler.org #393: Fehler unter Chrome bei 'export-Funktion'

UdoSchake, 2016-02-07:
Wenn ich unter Chrome versuche die Daten als CSV zu exportieren bekomme ich folgende Fehlermeldung:
[image]

Unter FF funktioniert es:
[image]

Gruß
Udo

andig, 2016-02-07:
Gerade nachgetestet- geht bei mir, Chrome 48 auf Win 64. Aktuelles git?

UdoSchake, 2016-02-07:
Jetzt aktuelles git.
Chrome 48 auf Win 64.
Cache und cookies gelöscht.
Fehler immer noch da.

Gruß
Udo

andig, 2016-02-07:
Kannst Dus bitte mal mit dem Demo Link oben probieren? Danach hilft nur noch TeamViewer- muss ich live sehen.

UdoSchake, 2016-02-07:
Also mit Chrome auf http://demo.volkszaehler.org/frontend funktioniert es.

andig, 2016-02-10:
@UdoSchake ich machs zu da nicht reproduzierbar. Wenns wieder auftaucht bitte bei mir melden.

closed 2016-02-10 by andig