volkszaehler/volkszaehler.org #396: Einheiten an den Achsen anzeigen

Rainer-Zufall, 2016-02-09:
Bitte fügt an den Achsen die Einheiten an, wenn die Werte Ähnlich sind ist eine Unterscheidung der Achsen fast nicht möglich.

Stelle mir das so vor das einfach m³ oder KWH angefügt wird. 
[image]

Danke!

andig, 2016-02-12:
Wie wärs damit:

[image]

Leider bekomme ich den Tick vor der Einheit nicht weg- sonst wärs eigentlich perfekt?!

bgewehr, 2016-02-12:
Ich finde es richtig gut so!

Rainer-Zufall, 2016-02-12:
Ja das sieht gut aus!
Evtl die Einheiten noch bis oben ziehen wie es früher war?

andig, 2016-02-12:
Gibt die Chartlib leider nicht her. Den letzten "tick" könnte ich noch weghacken, mehr geht leider nicht. Wäre mal gut mit negativen Achsenwerten zu testen...

Rainer-Zufall, 2016-02-12:
Dann bleibt es halt so. 👍
Mfg,

Sollner, 2016-02-14:
könnte man die Syntax in der options.js zu obigen Beispiel mal hier einstellen?
gerne auch erweitert um einen Max oder Min-Wert
Danke!

andig, 2016-02-14:
Welche Sytax wofür?

Sollner, 2016-02-15:
die entsprechenden Einträge für die Achsen in der options.js zu deinem obigen Beispiel

ps: 
deinen Ratschlag bezüglich Max-Wert bei Betriebsstundenzähler mag ich gerne umsetzen
nur was ist der Max-Wert wenn der Graph nur 0 und 1 annimmt
und der 1er-Wert dynamisch grafisch an meinen größten "anderen" Wert  angepasst wird
Danke.

andig, 2016-02-15:
> die entsprechenden Einträge für die Achsen in der options.js zu deinem obigen Beispiel

Gar keine. Das ist ist Coding und muss noch hübsch gemacht werden.

> deinen Ratschlag bezüglich Max-Wert bei Betriebsstundenzähler mag ich gerne umsetzen

Gut- hat aber auch mit diesem Issue gar nicht zu tun. Wäre doch schön die Themen nicht immer durcheinander zu würfeln....

Rainer-Zufall, 2016-02-26:
ist das jetzt schon implementiert?

andig, 2016-03-21:
> ist das jetzt schon implementiert?

Soeben via https://github.com/volkszaehler/volkszaehler.org/pull/398 implementiert und bereit zum Test. Falls Fehler bitte dieses Issue wieder öffnen.

closed 2016-03-21 by andig