volkszaehler/volkszaehler.org #410: aggregate sollte sich beenden, wenn es gerade schon läuft

justinotherguy, 2016-02-21:
Moin,
die Last auf meiner vz-Maschine ist gerade zu hoch (> 60).
Mir fällt auf, dass die Aggregate-Cron-Jobs jede Minute laufen und die Situation zumindest verschlimmern.
Drum:
Ideal wäre, wenn das aggregate-Skript zuerst prüfen würde, ob es gerade schon läuft.
Läuft es schon (noch), führt ein erneutes Ausführen zu einer (noch) höheren Last, drum sollte es sich (mit Ausgabe einer Fehlermeldung) beenden.

andig, 2016-02-21:
Was macht das denn jede Minute?! Pid prüfen wäre auch eher was für den cron job, nicht für php, oder?

justinotherguy, 2016-02-21:
naja - wie in http://wiki.volkszaehler.org/doku.php/howto/datenmengen empfohlen:
* * * * *  php /var/www/volkszaehler.org/misc/tools/aggregate.php run -m delta -l minute >/dev/null
Oder war Deine Frage anders gemeint?

justinotherguy, 2016-02-21:
Die PID zu prüfen ergibt m.E. keinen Sinn - bei jedem Aufruf ist das ja eine neue. Allerdings könnte man über einen Wrapper ein lock-file erzeugen...ich guck mal

justinotherguy, 2016-02-21:
ok, lockrun (http://unixwiz.net/tools/lockrun.html) scheint das passende Tool zu sein. Hab's eingerichtet.
Und den Wiki-Eintrag "Umgang mit den Datenmengen" http://wiki.volkszaehler.org/doku.php/howto/datenmengen hab ich auch gleich angepasst (vermutlich wäre das ein sinnvolles Default; ich lass das erst mal eine Weile laufen; mal sehen, ob es sich bewährt).
Ich habe übrigens die internen Links auf der Seite auch gleich angepasst; das waren externe Links (und die haben daher inzwischen nicht mehr funktioniert).
Und die anderen internen Links im Wiki auch...

andig, 2016-02-21:
Die Frage ist warum jede Minute- fallen da so viele Daten an dass es notwendig ist? Sonst vielleicht einfach alle 15min laufen lassen?

justinotherguy, 2016-02-21:
warum jede Minute? Weil das so im Wiki steht :)
Frage ist also: warum steht das so im Wiki?
Anders rum: wenn es seltener aufgerufen wird, hat er ja jeweils mehr zu tun, oder?
Seit ich das gg. 20:50 eingerichtet habe, kollidiert es jede volle Stunde mit dem "hourly"-Lauf.
Vielleicht habe ich das Konzept noch nicht verstanden:
- es gibt die 3 Aggregationsstufen (minute, hour, day)
- für jede gibt es eine eigene Aggregation
- je häufiger ist aggregate aufrufe, desto weniger gibt es pro Lauf zu tun
-> stimmt?
-> wie verringern wir die Wahrscheinlichkeit einer Kollision?
Ich könnte des "minute"-Lauf zu jeder vollen Stunde ausnehmen; das würde die bisherigen Kollisionen umgehen.

jahir, 2016-02-22:
"PID prüfen" heißt ja i.A., schauen ob es ein PIDFile gibt (/var/run/bla.pid oder so) und ob die darin stehende PID noch aktiv ist. Ist aber eher für daemons interessant, um die beenden oder reloaden zu können.
Für "exklusiv laufen" benutze ich flock, das ist im Paket util-linux und somit i.A. schon installiert:
```
flock --nonblock /tmp/vz-aggr.lock /dein/script
```
Das _--nonblock_ sorgt dafür, daß sofort beendet und nicht gewartet wird, bis der vorherige Lauf fertig ist.

@justinotherguy  Was hat du denn aktuell eingestellt? Ich gehe mal davon, daß die verschiedenen Aggregationsstufen unabhänig voneinander sind (hab's leider immer noch nicht geschafft, daß einzurichten), von daher würde ich auch einfach verschiedene lockfiles benutzen. Oder einfach nur den minute-Lauf locken, die anderen beiden laufen ja eh so selten, daß sie sich nicht in die Quere kommen sollten (sonst hast du eh ein Problem).
```
* * * * *  flock -n /tmp/vzagg.minute.lock php /var/www/volkszaehler.org/misc/tools/aggregate.php run -m delta -l minute >/dev/null
1 * * * *  php /var/www/volkszaehler.org/misc/tools/aggregate.php run -m delta -l hour >/dev/null
0 1 * * * php /var/www/volkszaehler.org/misc/tools/aggregate.php run -m delta -l day >/dev/null
```
Wobei ich es schon bedenklich finde, wenn die Aggregation einer Minute länger als eine Minute dauert. Das heißt ja letztendlich, daß die DB unter Dauerlast steht und keine Luft da ist...

justinotherguy, 2016-02-22:
aha - flock; hab das gerade mal abgeändert ("mit dabei" ist gewinnt natürlich!); mal sehen, wie das läuft.

> Was hat du denn aktuell eingestellt? Ich gehe mal davon, daß die verschiedenen 
> Aggregationsstufen unabhänig voneinander sind 
korrekt

> von daher würde ich auch einfach verschiedene lockfiles benutzen. Oder einfach nur 
> den minute-Lauf locken, die anderen beiden laufen ja eh so selten, daß sie sich nicht in die
> Quere kommen sollten (sonst hast du eh ein Problem).
jepp! bei hour und day sollte das nicht kollidieren (und wenn: kein Problem verursachen).

> Wobei ich es schon bedenklich finde, wenn die Aggregation einer Minute länger als eine Minute 
> dauert. Das heißt ja letztendlich, daß die DB unter Dauerlast steht und keine Luft da ist...
naja - meist dauert das weniger als eine Minute; wenn aber die Last gerade mal hoch ist, schaukelt sich das dann noch höher, weil die Aggregation ja dann noch dazu kommt - und das jede Minute erneut. Ich weiß nicht, wie die Aggregation intern arbeitet (s. meine Frage oben).
Ich vermute, dass der zweite Aggregationslauf nur von einem bereits abgeschlossenen ersten Lauf profitiert. Solange der erste noch läuft, macht der zweite vermutlich das Selbe. @andig: ist das so?

andig, 2016-02-22:
Ich verstehe nicht warum ein minütlicher lauf so lange braucht, insbesondere mehr als 10min. Entweder gibts da riesige Datenmengen oder furchtbar langsame queries die mir auf dem pc nicht aufgefallen sind.

Bin nicht sicher ob ich da schon sql logging eingebaut hatte, falls ja mal bitte mit laufen lassen, falls nein muss ich das einbauen.

Nebenbei: ich möchte die ganze sql aggregate logik gerne wegwerfen und durch die Gruppierung im Interpreter ersetzen. Damit liessen sich dann auch virtuelle Kanäle aggregieren und damit speichern. Für so einen Grossumbau fehlt momentan leider die Zeit.

justinotherguy, 2016-02-24:
> Ich verstehe nicht warum ein minütlicher lauf so lange braucht, insbesondere mehr als 10min

der Lauf alleine braucht nicht so lange; wenn aber gerade der "hour"-Lauf braucht der "minute"-Lauf länger als eine Minute; im Zweifel deaktiviere ich den "minute"-Lauf zur vollen Stunde.
Ansonsten kollidiert es nur mit dem Backup.

> Bin nicht sicher ob ich da schon sql logging eingebaut hatte, falls ja mal bitte mit laufen lassen

wie krieg ich das raus?

andig, 2016-02-24:
> der Lauf alleine braucht nicht so lange; wenn aber gerade der "hour"-Lauf braucht der "minute"-Lauf länger als eine Minute; im Zweifel deaktiviere ich den "minute"-Lauf zur vollen Stunde.
Ansonsten kollidiert es nur mit dem Backup.

Dann habe ich das Problem noch nicht verstanden. Wenn Du die Datenbank zuballerst ists ja kein Wunder. Ich würde wie gesagt auch nicht jede Minute aggregieren falls Du nicht im Millisekundenraster aufzeichnest. Sollte völlig reichen das entweder ga nicht oder nur alle 10min, dann ausserhalb der vollen Stunde, laufen zu lassen.

justinotherguy, 2016-02-24:
...dann versuch ich's nochmal: :)
- der Grund, weshalb ich den "minute"-Lauf jede Minute starte ist: so steht es im Wiki
- wenn Du sagst "alle 10 Minuten; am Besten immer um *5 Minuten ist perfekt", trage ich das so bei mir und im Wiki ein
- ist das aber dann nicht so, dass es dann eben 10x so viel zu tun gibt?
- was Du mit "Wenn Du die Datenbank zuballerst ists ja kein Wunder." meinst, verstehe ich nicht; die DB wird ja nicht von mir alleine, sondern von weiteren Usern genutzt (demo.vz eben)

Jetzt besser? :-)

andig, 2016-03-11:
> der Grund, weshalb ich den "minute"-Lauf jede Minute starte ist: so steht es im Wiki

Dann würde sich anbieten das zu ändern ;)

> wenn Du sagst "alle 10 Minuten; am Besten immer um *5 Minuten ist perfekt", trage ich das so bei mir und im Wiki ein

Naja- ist ein klares "nach Bedarf". Ich logge alle 5min, hab also gar keine Aggregation auf Minuten laufen...

> ist das aber dann nicht so, dass es dann eben 10x so viel zu tun gibt

Klar. Aber die Setupkosten (einmalig die Daten identifizieren) bleiben gleich. Und s.o.- 10x muss ja kein Problem sein.

Mein Vorschlag: beobachten und anpassen, Issue schließen.

closed 2016-03-21 by andig