volkszaehler/volkszaehler.org #415: Code: Improve interpreter naming

andig, 2016-03-24:
The current interpreters (https://github.com/volkszaehler/volkszaehler.org/tree/master/lib/Interpreter) naming is highly confusing. Need to check what they actually do every single time I need to change things:

  - Sensor reads current values (e.g. Leistung)
  - Meter counts impulses (e.g. S0) while a meter is really a "Zähler"
  - Counter does not count but takes actual meter readings in terms of "Zählerstände" (e.g. Verbrauch in kWh)

Suggestion to rename:

  - `SensorInterpreter` -> `SensorInterpreter` (as is)
  - `MeterInterpreter` -> `ImpulseInterpreter` (good description of what it does)
  - `CounterInterpreter` -> `ReadingInterpreter` (doesn't make me entirely happy but seems to be the best translation around)

In any case Sensor + Impulse are obvious and therefore make clear what Reading does. This is a potentially breaking change for anyone who has defined there own entity types.

Please comment.

jahir, 2016-03-26:
I'm not fond of changing names, but IMO this is purely an internal change. Anybody using their own entity type is probably able to fix this by themself.
Go for it.

andig, 2016-06-25:
Ideas: call the `CounterInterpreter` a `Sum(mator)Interpreter` or `MeterReadingInterpreter` (very similar to existing `MeterInterpreter`) or `DialInterpreter` or `CyclometerInterpreter` (Rollenzählwerk).

`AccumulatorInterpreter`?!

PR #446 (2016-07-04): Improve interpreters internal naming
Fix #415 

**Breaking change** the naming of the Interpreters in `EntityDefinition.json` changes. This will affect *anybody* who has created custom entity types:

  - `SensorInterpreter` -> `SensorInterpreter` (as is)
  - `MeterInterpreter` -> `ImpulseInterpreter`
  - `CounterInterpreter` -> `AccumulatorInterpreter`
closed 2016-07-04 by andig