volkszaehler/volkszaehler.org #423: Frontend: ability to abort zoom-in triggered by click-drag-release

mklhmnn, 2016-04-10:
- all click-drag-release events should be ignored if the time between click and release is less then, e.g. 1s
- click-drag should abort either on pressing ESC or when moving the mouse vertically outside the view

andig, 2016-04-10:
Whats the use case? Is there an issue with the current behaviour?

bgewehr, 2016-04-10:
Since drag and drop is used to build structured groups of channels this requirement should be rethought.

mklhmnn, 2016-04-10:
The problem is that, especially with a touchpad, it is possible to trigger the click-drag-release by accident when trying to move the cursor. Undoing this zoom-in is not very easy to achieve.

andig, 2016-04-10:
I'm confused. Zoom out is the minus button. What exactly are you trying to achieve?

mklhmnn, 2016-04-10:
I want to prevent a) tiny drags (zoom into a very tiny time period - hard to undo with a lot of minus button presses and scrolling) and b) larger drags (on any operating system you can abort drag and drop operations by pressing ESC; it's for a good reason).

andig, 2016-05-17:
Still unclear for me, sorry.

closed 2016-05-31 by andig