volkszaehler/volkszaehler.org #424: Frontend: ability to specify a page parameter to show a certain last time period

mklhmnn, 2016-04-10:
Maybe it is already there, but at least it is not obvious: I want to bookmark a link to the frontend that always shows me the latest, e.g., 2 hours.

andig, 2016-04-10:
Please check mailing list. The topic has been discussed this week.

closed 2016-04-10 by andig

mklhmnn, 2016-04-10:
Why closed? According to the topic http://article.gmane.org/gmane.network.volkszaehler.user/6440 it looks like it is not supported right now.

andig, 2016-04-10:
Exactly, but there's a perfect workaround and explanation on the mailing list.

mklhmnn, 2016-04-10:
I've tried now that (I'm a PHP noob):

<?php
	$period = 3 * 3600;
	if (isset($_GET['period'])) {
	    $period = $_GET['period'];
	}
	date_default_timezone_set("Europe/Berlin");
	$now = time();
	$last = $now - $period;
	$yesterday = date("Y-m-d", strtotime("now"));
	$from_time = date("Y-m-d", $last)."T".date("H:i", $last).":00";
	$to_time = date("Y-m-d", $now)."T".date("H:i", $now).":59";
	header ("Location: /frontend/?from=$from_time&to=$to_time");
?>

Maybe there is a better solution? Something like that could be shipped by default or supported directly?