volkszaehler/volkszaehler.org #428: Issue with push server

Sineos, 2016-04-23:
I cannot get the push server function to work. This is my setup:

IP: 192.168.178.10
- vzlogger (Commit: 9ce42e3)
- Input of 4x S0, 3 SML channel
- [Config](http://pastebin.com/bZk7U0cW)

IP: 192.168.178.8
- volkszaehler.org (Commit: 0a75e7a)
- [Config](http://pastebin.com/BdKXZhrz)

With the volkszeahler frontend and middleware this setup works nicely and without any problems.

**Issue Scenario 1:**
Now I want to start the push-server on 192.168.178.8. It starts up fine, but never receives any data. Looking at the network connections it seems that no connection is established between vzlogger (IP 192.168.178.10) and the push-server (IP 192.168.178.8)

On 192.168.178.10
>  # netstat -an |grep 5582
>  tcp        0      0 192.168.178.10:59071    192.168.178.8:5582      CLOSE_WAIT

On 192.168.178.8
> #netstat -an|grep 5582
> tcp        0      0 0.0.0.0:5582            0.0.0.0:*               LISTEN
> tcp        0      0 192.168.178.8:5582      192.168.178.10:59221    TIME_WAIT
> tcp        0      0 192.168.178.8:5582      192.168.178.10:59224    TIME_WAIT
> tcp        0      0 192.168.178.8:5582      192.168.178.10:59294    TIME_WAIT
> tcp        0      0 192.168.178.8:5582      192.168.178.10:59194    TIME_WAIT
> ...

**Issue Scenario 2:**
For testing purposes I switched the push config in vzlogger back to localhost and started the push server purely on the vzlogger server:

>  php volkszaehler.org/misc/tools/push-server.php
> Volkszaehler push server
> Listening for updates on 5582. Clients may connect at 8082.
> Trying to subcribe to invalid topic 1bb99c20-0595-11e6-af6f-7b9c7a908836
> Trying to subcribe to invalid topic dabe49b0-05a0-11e6-82c7-3d516e19bf01
> Trying to subcribe to invalid topic 9ff4a200-0599-11e6-9d64-dd628ac55cb4
> Trying to subcribe to invalid topic dabe49b0-05a0-11e6-82c7-3d516e19bf01
> Trying to subcribe to invalid topic 1bb99c20-0595-11e6-af6f-7b9c7a908836

So, here the push-server establishes a connection with vzlogger, but now is claiming that the topics are invalid. Strange thing is that the very same topics are properly shown in the frontend.

Maybe I have messed something up, but now I'm out of ideas. Any help would be appreciated.

andig, 2016-04-24:
> Scenario 1:
> Now I want to start the push-server on 192.168.178.8. It starts up fine, but never receives any data. Looking at the network connections it seems that no connection is established between vzlogger

Did you check if you can manually connect to the push server at 192.168.178.8:5582? Also, what does the vzlogger log file have to say?

> Scenario 2:
> For testing purposes I switched the push config in vzlogger back to localhost and started the push server purely on the vzlogger server

That won't work as push-server needs the middleware to convert vzlogger readings using the various interpreters. It does therefore need to run on the same machine as the middleware for the time being.

Sineos, 2016-04-24:
Thanks for the pointers but unfortunately it seems, that I do not understand the concept.

Here is where I am right now:

The vzlogger.log indicates that the push is indeed send:

> [Apr 24 22:01:38][push] push: { "data": [ { "uuid": "9ff4a200-0599-11e6-9d64-dd628ac55cb4", "tuples": [ [ 1461528098234, 660.90000000000009 ] ] } ] }
> [Apr 24 22:01:38][push] added to uuid 9ff4a200-0599-11e6-9d64-dd628ac55cb4
> [Apr 24 22:01:38][chn1] Adding reading to queue (value=6572687.90 ts=1461528098233)
> [Apr 24 22:01:38][push] added to uuid bcbaa290-05a0-11e6-840a-8ba5363bf363
> [Apr 24 22:01:38][chn2] Adding reading to queue (value=7542122.70 ts=1461528098233)
> [Apr 24 22:01:38][push] added to uuid dabe49b0-05a0-11e6-82c7-3d516e19bf01
> [Apr 24 22:01:38][push] CURL Request to http://192.168.178.8:5582 succeeded with code: 200
> [Apr 24 22:01:38][push] send ok to url http://192.168.178.8:5582
> [Apr 24 22:01:38][push] push: { "data": [ { "uuid": "dabe49b0-05a0-11e6-82c7-3d516e19bf01", "tuples": [ [ 1461528098233, 7542122.7000000002 ] ] }, { "uuid": "bcbaa290-05a0-11e6-840a-8ba5363bf363", "tuples": [ [ 1461528098233, 6572687.9000000004 ] ] } ] }

After modifying volkszaehler/htdocs/frontend/javascripts/options.js to uncomment the live port (http://pastebin.com/ShKf9pqx), a _tcpdump -i eth0 -vvvvtttAXns 1500 'port 8082'_ on the volkszaehler box (IP 192.168.178.8) indicates that the push is received and send directly to the frontend for visualization:

> 00:00:01.631187 IP (tos 0x0, ttl 64, id 30619, offset 0, flags [DF], proto TCP (6), length 208)
>     192.168.178.8.8082 > 192.168.178.11.54228: Flags [P.], cksum 0xe627 (incorrect -> 0x2a65), seq 10067:10235, ack 1, win 245, length 168
>         0x0000:  4500 00d0 779b 4000 4006 dd27 c0a8 b208  E...w.@.@..'....
>         0x0010:  c0a8 b20b 1f92 d3d4 66d3 898a 0299 87c1  ........f.......
>         0x0020:  5018 00f5 e627 0000 817e 00a4 5b38 2c22  P....'...~..[8,"
>         0x0030:  3966 6634 6132 3030 2d30 3539 392d 3131  9ff4a200-0599-11
>         0x0040:  6536 2d39 6436 342d 6464 3632 3861 6335  e6-9d64-dd628ac5
>         0x0050:  3563 6234 222c 227b 5c22 7665 7273 696f  5cb4","{\"versio
>         0x0060:  6e5c 223a 5c22 302e 335c 222c 5c22 6461  n\":\"0.3\",\"da
>         0x0070:  7461 5c22 3a7b 5c22 7575 6964 5c22 3a5c  ta\":{\"uuid\":\
>         0x0080:  2239 6666 3461 3230 302d 3035 3939 2d31  "9ff4a200-0599-1
>         0x0090:  3165 362d 3964 3634 2d64 6436 3238 6163  1e6-9d64-dd628ac
>         0x00a0:  3535 6362 345c 222c 5c22 7475 706c 6573  55cb4\",\"tuples
>         0x00b0:  5c22 3a5b 5b31 3436 3135 3239 3937 3732  \":[[14615299772
>         0x00c0:  3731 2c36 3136 2e33 2c31 5d5d 7d7d 225d  71,616.3,1]]}}"]

The IP 192.168.178.11 is the box that had the frontend opened during the dump.

With the above modification of the options.js, the frontend is suddenly displaying an error message upon refresh:

> uncaught exception: protocol error (welcome message received more than once)

But sofar this seems to work as the frontend is continuously updating the values.

Now my initial intend was to wire up a node-red flow to catch the readings and push it to a mqtt broker. Here I'm currently failing completely. Regardless what Input-Node I'm trying, I do not receive a single value from the push server. I tried TCP, Websocket and HTTP. At least with TCP or Websocket I can see, that a connection is made to the 8082 port but no data is received.

andig, 2016-04-25:
> The vzlogger.log indicates that the push is indeed send:

ok

> Did you check if you can manually connect to the push server at 192.168.178.8:5582? 

Still not answered, but

> With the above modification of the options.js, the frontend is suddenly displaying an error message upon refresh:

Apparently you get values in the frontend, so the connection from vzlogger to push server ist working. Good.

> With the above modification of the options.js, the frontend is suddenly displaying an error message upon refresh:

Where is that error? Popup or dev console? Could be local firewall or similar.

> Now my initial intend was to wire up a node-red flow to catch the readings and push it to a mqtt broker. Here I'm currently failing completely. Regardless what Input-Node I'm trying, I do not receive a single value from the push server. I tried TCP, Websocket and HTTP. 

None of those will work. The push server uses WAMP- I'm not aware of a WAMP node for node-red.

> At least with TCP or Websocket I can see, that a connection is made to the 8082 port but no data is received.

WAMP is a protocol using websockets. Why don't you- instead of connecting to the push server- ask vzlogger to push directly to node-red? You'd get "raw" meter values though, not the ones transformed by the middleware.

Sineos, 2016-04-25:
> Where is that error? Popup or dev console? Could be local firewall or similar.
This is a popup in the browser that is displaying the live values. It appeared after modifying options.js to uncomment the live port. In addition it is only displayed after hitting "refresh" on the frontend. After clicking OK on the error popup, the frontend seems to be working as intended.

> WAMP is a protocol using websockets. Why don't you- instead of connecting to the push server- ask vzlogger to push directly to node-red? You'd need an HTTP end point for that, but you'd get "raw" meter values though, not the ones transformed by the middleware. 
Ok, now this is a pity. I had hoped I can get "ready to use" values. Especially for the S0 values, this would have been really nice.
Are there any plans from your side to support this? A dedicated volkszaehler-node would be a treat ;-)

andig, 2016-05-02:
> Are there any plans from your side to support this? A dedicated volkszaehler-node would be a treat 

Yes there are, but thats a totally separate topic than your initial request, right? I'll change the title accordingly.

andig, 2016-05-20:
Closed via #430

closed 2016-05-20 by andig