volkszaehler/volkszaehler.org #431: Datenbereichsauswahl per Maus wird abgebrochen bei "Dateneingang"

schnello, 2016-05-17:
Die Selektion per Maus über einen Bereich wird abgebrochen sobald im Frontend neue Daten dargestellt werden. Bei Livedaten aus dem pushserver (Intervall ca 1 Sekunde) ist somit eine Auswahl per Maus nur mehr schwer möglich.

bgewehr, 2016-10-23:
Another isue, same reason: 
Each time i was hovering the chart with my mouse i got values shown in the legend.
Since i use the push server they get lost every second.
Can't we remember i was hovering and keep that state even if the chart moves on due to received push values?
If not, the hovering mode has become useless and we need another possibility to read values from a specific chart time. Maybe set a time mark which stays constant even if the chart moves on and display the values of that time mark in the legend?

andig, 2016-10-23:
Leider steht Pushserver grad gaaaanz unten auf meiner Liste- erstmal muss ich nen Raspi3 beschaffen, dann neu installieren, dann vzlogger in Betrieb nehmen und mein Setup umziehen und dann kann ichs anschauen. Sorry...

christian-weiss, 2017-12-20:
Bug ticket is more then 1 year old: Is this bug still existing?

andig, 2018-03-29:
ping @schnello @frankrichter @bgewehr würdet ihr das mal testen?

schnello, 2018-03-29:
Hallo.

Habs getestet und konnte leider keine Veränderung feststellen.

Grüße

frankrichter, 2018-03-29:
Hier klappt es leider auch nicht.

frankrichter, 2018-03-29:
Update: ohne die Klammern hinter `vz.plot.getSelection` funktioniert's.

andig, 2018-03-29:
> Update: ohne die Klammern hinter vz.plot.getSelection funktioniert's.

Schwer vorstellbar bzw. Zufall - dann wird nämlich gar keine Funktion aufgerufen. Ok, schaue ich nochmal an :/

frankrichter, 2018-03-29:
War Unfug , sorry. Das blockiert Push-Updates generell...

frankrichter, 2018-03-29:
Wenn ich die Erläuterung in volkszaehler.org/htdocs/vendor/flot/jquery.flot.selection.js richtig verstehe, wird getSelection erst nach erfolgreicher Selektion (Maustaste wieder losgelassen) ein Wert zugewiesen:

> getSelection()
> Returns the current selection in the same format as the "plotselected"
> event. If there's currently no selection, the function returns null.

Für unsere Zwecke könnte das hilfreich sein:

> A "plotselecting" event is fired during the process with the same
> parameters as the "plotselected" event, in case you want to know what's
> happening while it's happening

andig, 2018-03-31:
Bitte nochmal schauen- die relevanten Teile sind eigentlich nur die paar Zeilen in `wui.js`. Bei mir lokal tut das. Problem war einfach dass der Redraw nicht passiert wenn die Daten kommen sondern erst nach Delay- und genau da wurde dann nicht mehr auf eine Selektion geprüft.

closed 2018-03-31 by andig

andig, 2018-04-02:
@bgewehr separates Issue

PR #684 (2018-03-31): Don't update chart when selection is active
Fix #431