volkszaehler/volkszaehler.org #432: Livedaten von einem Meter mit Zählerständen

schnello, 2016-05-17:
Live Darstellung im Frontend von Daten eines Zählerstand-Meters ist "unschön" (siehe screenshot)

[image]

andig, 2016-05-19:
Ich habe PR https://github.com/volkszaehler/volkszaehler.org/pull/430 nochmal akualisiert. Bitte schau mal ob commit https://github.com/andig/volkszaehler.org/commit/c7d4c975b1c666290fac3823aa080cdb8989a21b das Problem löst.

schnello, 2016-05-19:
Ist leider unverändert.

andig, 2016-05-19:
Mhm. Zeigt Dir git log dass der Commit wirklich drin ist?

schnello, 2016-05-19:
Ich bin jetzt nicht der git profi aber ich würde meinen ich hab alles richtig gemacht.

```
root@nas:/var/www/volkszaehler.org# git branch -v
  master 0a1363d [behind 4] Fix hhvm travis error
  test   44e0039 Merge commit 'refs/pull/430/head' of https://github.com/volkszaehler/volkszaehler.org into test
* test2  c7d4c97 Suppress duplicate counter values
```

Bzgl ssh:
Schick mir bitte per Mail deinen öffentlichen Key per Mail.

Grüße

schnello, 2016-05-20:
Nachtrag:
Ich glaub ich hab den Fehler auf meiner Seite gefunden. Ich habe den Pushserver selbst nicht neugestartet. jedoch kann ich diesen nun gar nicht mehr starten:
`
Volkszaehler Push Server
Listening for updates on 5582. Clients may connect at 8082.
PHP Catchable fatal error:  Argument 1 passed to Volkszaehler\Server\MiddlewareAdapter::addAdapter() must be an instance of Volkszaehler\Server\PushClientMessageInterface, i
nstance of Volkszaehler\Server\WampClientAdapter given, called in /var/www/volkszaehler.org/misc/tools/push-server.php on line 97 and defined in /var/www/volkszaehler.org/li
b/Server/MiddlewareAdapter.php on line 62
root@nas:/home/nas# 
`

andig, 2016-05-20:
Yep, habs gesehen und bin dran.

andig, 2016-05-20:
@schnello versuchs nochmal

schnello, 2016-05-20:
Sieht gut aus

andig, 2016-05-20:
Closed via https://github.com/volkszaehler/volkszaehler.org/pull/430

closed 2016-05-20 by andig