volkszaehler/volkszaehler.org #436: Wunsch - min/max zulässige Werte pro Kanal

kr0815, 2016-05-25:
Leider gibt es Sensoren die sporadisch falsche Werte senden, so z.B. meine Heizungssteuerung statt 53.8°C 128°C - das gibt dann unschöne Peaks im Diagramm

Man könnte das jetzt im sendenden Device unterbinden, leider ist das nicht immer einfach möglich.
(Ich setze kein VZLogger ein)

deshalb die Idee - man gibt pro Kanal optional min/max-Werte vor, alles darüber / darunter wird einfach ignoriert ?

andig, 2016-05-25:
M.E. wäre das falsch. Die MW speichert und analysiert- für korrekte Eingangsdaten sind die Geräte/Controller verantwortlich. Wenn Du keinen vzlogger nutzt ohnehin leicht zu realisieren.

andig, 2016-05-31:
@kr0815 Du kannst das mit wenig Aufwand aber im DataController selbst realisieren.

closed 2016-05-31 by andig