volkszaehler/volkszaehler.org #442: group parameter und bezogene Energie

Defibrillat0r, 2016-05-30:
Hallo

Mit dem 'group'-Parameter kann ich ja die abgefragten Werte zeitlich gruppieren. Werden mir hier die durchschnittlichen Leistungen über die angegebene Dauer(Stunde, Tag etc) sowie der aktuelle Wert zurückgegeben? z.B. [1]

Weiter möchte ich ein Balkendiagramm darstellen, welches die bezogene Energie zum Beispiel pro Tag über eine Woche darstellt (Mo: 25kWh, Di: 80kWh etc.). Gibt es hierfür eine ähnliche Funktion wie der group parameter? Wäre eine super Funktionalität auch wenn ich hier gelesen habe, dass das aufwändig zum Implementieren wäre. Ist hier etwas in Planung?
Ansonsten müsste ich für jeden Tag, der dargestellt werden soll, eine Abfrage machen und die consumption zwischenspeichern, oder?

Falls ich ein paar Inputs bekomme, wo/wie die Funktionalität eingebaut werden kann (zur Orientierung), könnte ich etwas Zeit investieren.

[1] http://demo.volkszaehler.org/middleware.php/data/c673b290-fdac-11e0-a470-1d9351203a00.json?from=80+days+ago&to=now&group=day

andig, 2016-05-30:
Für Diskussionen die Mailingliste. Deine Frage habe ich nicht konkret verstanden. Falls Du sowas meinst: ja ist in Planung:

[image]

Defibrillat0r, 2016-05-30:
Ja. Sowas mein ich.
Gibt es dazu mehr Informationen? "Roadmap"? Ein zugehöriger Commit? Wann ist damit zu rechnen? Information für contribution?

andig, 2016-05-31:
Die Roadmap steht letztlich hier im repository in Form der offenen Enhancements. ETA ist wie immer- wenns fertig ist.

andig, 2016-06-25:
Todo: disable live updates when in consumption mode.

andig, 2017-02-06:
I have a partial solution in my repo at https://github.com/andig/volkszaehler.org

andig, 2019-01-24:
Not sure what you‘re asking here, but daily will only start working after one day since it needs one timestamp before the current period. May not be intuitive :/

closed 2019-03-21 by andig