volkszaehler/volkszaehler.org #447: composer: "lock file is not up to date"

J-A-U, 2016-06-29:
[Ein Nutzer](http://www.photovoltaikforum.com/volkszaehler-org-f131/wie-aktualisiere-ich-meine-middleware--t113237.html#p1399937) ist über die Warnung gestolpert:
> Warning: The lock file is not up to date with the latest changes in composer.json. You may be getting outdated dependencies. Run update to update them.

Da muss wohl die composer.lock mal aktualisiert werden:
`$ composer update --lock`

andig, 2016-06-29:
Die Antwort findet sich hier: http://stackoverflow.com/questions/33052195/what-are-the-differences-between-composer-update-and-composer-install

Für updates ist es daher besser composer update statt composer install zu nutzen, muss ggf im Wiki aktualisiert werden.

J-A-U, 2016-06-29:
Wiki ändern - Check.

andig, 2016-07-04:
Können wir das Issue dann wieder zu machen?

closed 2016-07-05 by J-A-U