volkszaehler/volkszaehler.org #448: Total value is way too high

hundsmiachn, 2016-07-15:
Hi

I'm having a small issue with the values at the bottom. When I select the year view it says a total usage of 1.1 MWh. All other views (month, day,..) show the correct value.

[image]

regards
Erich

andig, 2016-07-15:
Hallo Erich,

Laut Screenshot ca. 800W x 24h x 2 Monate (60 Tage) sind ca. 1100kWh also 1,1 MWh. Wo siehst Du den Fehler?

Viele Grüße, Andreas

hundsmiachn, 2016-07-15:
Hallo, sorry das stimmt natürlich. Ich hab mich da vertan. Mwh hat sich so viel angehört, aber es stimmt natürlich 
Sorry nochmals. 

Mfg
Erich

closed 2016-07-15 by hundsmiachn

andig, 2016-07-15:
Kein Problem- solte das für Strom mal in kWh ändern...