volkszaehler/volkszaehler.org #454: Idee: Einzelne Verbraucher erkennen und verfolgen

bgewehr, 2016-08-21:
So wie bei Smappee könnte auch Volkszaehler jedem Gerät mit spezifischem Einschaltstrombedarf oder Nutzungsmuster einen Unterkanal zuordnen damit den Verbrauch einzelner Geräte bestimmen und verfolgen.

http://www.t-online.de/digital/id_71498798/smappee-im-test-stromverbrauch-bis-ins-detail-messen-fuer-jedermann.html

Was meint Ihr? Gute Sache?

mbehr1, 2016-08-21:
Wäre eine super Sache.

Heißt in der Theorie „Power disaggregation“ und ist ein recht schwieriges Thema. Wichtig für eine einigermaßen gute Erkennung ist eine hohe Sampling-Frequenz. Ich nutze bei mir 1s als Takt. Das scheint noch leicht zu wenig zu sein. Kommerzielle Geräte und theoretische Analysen scheinen eher Frequenzen im Bereich von >10Hz (100Hz) zu nutzen.

> So wie bei Smappee könnte auch Volkszaehler jedem Gerät mit spezifischem Einschaltstrombedarf oder Nutzungsmuster einen Unterkanal zuordnen damit den Verbrauch einzelner Geräte bestimmen und [...]

Gruß

Matthias

andig, 2016-08-25:
Thema für die Mailinglist, kein Issue.

closed 2016-08-25 by andig