volkszaehler/volkszaehler.org #461: IE11: Javascript Runtime Error: Unbekannter Fehler

BlueAndi, 2016-09-06:
Ich habe ein Update per git pull gemacht und ebenso den composer aktualisiert. Nun bekomme ich im Frontend einen Dialog mit dem 

> Javascript Runtime Error: Unbekannter Fehler

.
Frontend läuft auf einem Raspberry Pi, mit Raspbian wheezy. Das Betriebsssytem ist auf dem neuestem Stand.

Das Problem habe ich mit dem Firefox v48.0.2 nicht. Nur beim IE v11.
Cookies und Cache habe ich gelöscht, ohne Erfolg.

J-A-U, 2016-09-06:
Aktiviere mal die Debug-Konsole:
Oben rechts aufs Zahnrad und "F12 Entwicklertools".

Als erstes wäre interessant ob im Reiter "Konsole" weiterführende Informationen zum Fehler zur Verfügung stehen.

BlueAndi, 2016-09-06:
[image]
[image]

J-A-U, 2016-09-06:
Jupp, kann ich bestätigen. Allerdings macht der IE bei mir (bei abgestelltem Debug) trotzdem weiter und zeigt alles richtig an.

BlueAndi, 2016-09-06:
Bei mir kommt immer dieser Dialog und wenn ich den mit "ok" bestätige, gehts auch weiter.
[image]

BlueAndi, 2016-09-06:
Ich habe mal kurz die aktuelle autobahn.js eingespielt. Der Fehler ist zwar weg, dafür kommt einer in init.js. Vermute da stimmen dann die Abhängigkeiten nicht mehr.

BlueAndi, 2016-09-06:
Hier die Ausgabe der Konsole bei Firefox, vielleicht hilft das weiter:
[image]

andig, 2016-09-07:
> Ich habe mal kurz die aktuelle autobahn.js eingespielt

Keine Ahnung was Du meinst. Die "aktuelle" ist bereits enthalten, also bitte rückgängig machen.

> Hier die Ausgabe der Konsole bei Firefox, vielleicht hilft das weiter:

Ich sehe da keinen Fehler. Anscheinend hast Du den Push Server nicht installiert, also alles wie erwartet. 

@BlueAndi Blöd ist der Fehler im IE. Kann ich mir das live anschauen? Url gerne per pm.

andig, 2016-09-07:
Der Fehler ist behoben

BlueAndi, 2016-09-07:
Wollte Dir gerade die URL freigeben. ;-)
Ich prüfe es gleich mal.

BlueAndi, 2016-09-07:
Alles perfekt. Keine Fehlermeldung mehr im IE11. Firefox habe ich auch kurz geprüft, sieht gut aus.

Was noch kommt ist folgende Meldung in der Konsole:

> SCRIPT12008: WebSocket Error: Incorrect HTTP response. Status code 404, Not Found

Vermute aber das liegt dann wieder am nicht installierten Push Server.

Vielen Danke für den echt superschnellen Support!!!

closed 2016-09-07 by BlueAndi