volkszaehler/volkszaehler.org #474: Hottracking Bug

milefork, 2016-09-19:
Moin,

das Hottracking (mit der Maus über die Datenreihen fahren) geht nicht mehr wenn vorher die Optionen ausgeklappt werden.

Gruß

andig, 2016-09-19:
Kannst Du rausfinden seit welchem commit (git bisect)?

milefork, 2016-09-19:
könnte ich bestimmt, bekomme aber bei den rollbacks den bug nicht weg:

`PHP Fatal error:  Class 'Volkszaehler\\Interpreter\\AccumulatorInterpreter' not found in /var/www/blah/lib/Controller/DataController.php on line 68`

Vielleicht schreibst du mich mal übers irc an, idle gerade in #volkszaehler.org auf freenode  [~fork@unaffiliated/fork]

andig, 2016-09-19:
Fehler gefunden. Spannend ist: das Gleiche passiert auch beim Ändern der Fenstergröße. Tatsächlich sollte das Frontend sich daran anscheinend aber dynamisch anpassen (ggf auch in der Höhe). Das hat es allerdings wohl noch nie getan...

andig, 2016-09-19:
Schau mal bitte ob der Fix passt. Nebenbei habe ich auch die bisher nur halb implementierte Funktion für Resize des Plots nachgerüstet.

Bisher auf keinem Mobilgerät getestet!

milefork, 2016-09-19:
@andig schaut gut aus! 
Funktioniert jetzt und auf dem Handy sieht die resized UI auch deutlich besser aus.

closed 2016-09-20 by andig