volkszaehler/volkszaehler.org #477: Elec. energy (S0 impulse) not shown in diagram

BlueAndi, 2016-09-19:
Did a fresh installation on a raspberry jessie and the current volkszaehler.org head.
The channel table shows some (seems to be constant) values and the diagram shows nothing.

The database receives values, I can see them with phpmyadmin.

J-A-U, 2016-09-19:
If the Raspi couldnt connect to a timeserver the date will be wrong and the timestamps of the data-tubles far in the past.

http://wiki.volkszaehler.org/howto/raspberry_pi_image "Einrichten"

BlueAndi, 2016-09-20:
Yes, you are absolut right. The timestamp looks bad. I was confused, because other values were shown correct, but thats because they were sent from a different device to the raspi.

closed 2016-09-20 by BlueAndi